motioneye-project/motioneye #429: Motion Detection on 1 camera; not another

mvh0518, 2017-01-21:
Motion detection works pretty well on one camera, but doesn't work at all on another, network camera.
On the non-motion-detecting camera, the red box is not even displayed on the motion eye web monitor page.  The non-motion-detecting camera is a network camera.  I had thought I had this camera detecting
motion in the past, but somehow I lost the correct configuration.
Attached is a diff of the motion.config files for the two cameras.  Beyond 'text_double', I see no significant differences between the two configurations.
What's wrong?
In the below, 'Office' works, indicated in diff by '<'
while 'Living Room' does not detect motion, indicated in diff by '>'
12c12
< # @id 2
---
> # @id 4
19c19
< # @name Office
---
> # @name Living Room
30d29
< height 240
32c31
< threshold 1305
---
> threshold 1843
39,40d37
< hue 0
< saturation 0
45c42
< movie_filename %Y-%m-%d/Office-%H-%M-%S
---
> movie_filename %Y-%m-%d/LRoom-%H-%M-%S
47,49c44,46
< stream_port 8082
< rotate 0
< brightness 0
---
> stream_port 8084
> rotate 180
> stream_auth_method 0
55c52
< stream_auth_method 0
---
> minimum_motion_frames 20
58c55
< text_double off
---
> text_double on
63,64c60,61
< text_left Office
< picture_filename %Y-%m-%d/Office-%H-%M-%S
---
> text_left Living Room
> picture_filename %Y-%m-%d/LRoom-%H-%M-%S
67,68c64
< contrast 0
< videodevice /dev/video0
---
> netcam_keepalive on
73c69
< ffmpeg_bps 44544
---
> ffmpeg_bps 43008
75,76c71,72
< minimum_motion_frames 20
< width 320
---
> netcam_tolerant_check on
> netcam_url http://192.168.1.147:8080/video

chrisf4lc0n, 2017-01-28:
Can you edit your post, so we can actually see something, not just the enormous letters?

mvh0518, 2017-01-28:
Sorry about  the fonts.  Here is the diff; thread 2 works thread 4 doesn't.
Thread 2 is "<" in diff; Thread 4 is ">".
$ diff thread-2.conf thread-4.conf
12c12
< # @id 2
---
> # @id 4
19c19
< # @name Office
---
> # @name Living Room
30d29
< height 240
32c31
< threshold 1305
---
> threshold 1843
39,40d37
< hue 0
< saturation 0
45c42
< movie_filename %Y-%m-%d/Office-%H-%M-%S
---
> movie_filename %Y-%m-%d/LRoom-%H-%M-%S
47,49c44,46
< stream_port 8082
< rotate 0
< brightness 0
---
> stream_port 8084
> rotate 180
> stream_auth_method 0
55c52
< stream_auth_method 0
---
> minimum_motion_frames 20
58c55
< text_double off
---
> text_double on
63,64c60,61
< text_left Office
< picture_filename %Y-%m-%d/Office-%H-%M-%S
---
> text_left Living Room
> picture_filename %Y-%m-%d/LRoom-%H-%M-%S
67,68c64
< contrast 0
< videodevice /dev/video0
---
> netcam_keepalive on
73c69
< ffmpeg_bps 44544
---
> ffmpeg_bps 43008
75,76c71,72
< minimum_motion_frames 20
< width 320
---
> netcam_tolerant_check on
> netcam_url http://192.168.1.147:8080/video

mvh0518, 2017-01-28:
Sorry, Markdown was messing up my post.  Took me a few minutes to find the magic to disable Markdown.  
<pre>
$ diff thread-2.conf thread-4.conf
12c12
< # @id 2
---
> # @id 4
19c19
< # @name Office
---
> # @name Living Room
30d29
< height 240
32c31
< threshold 1305
---
> threshold 1843
39,40d37
< hue 0
< saturation 0
45c42
< movie_filename %Y-%m-%d/Office-%H-%M-%S
---
> movie_filename %Y-%m-%d/LRoom-%H-%M-%S
47,49c44,46
< stream_port 8082
< rotate 0
< brightness 0
---
> stream_port 8084
> rotate 180
> stream_auth_method 0
55c52
< stream_auth_method 0
---
> minimum_motion_frames 20
58c55
< text_double off
---
> text_double on
63,64c60,61
< text_left Office
< picture_filename %Y-%m-%d/Office-%H-%M-%S
---
> text_left Living Room
> picture_filename %Y-%m-%d/LRoom-%H-%M-%S
67,68c64
< contrast 0
< videodevice /dev/video0
---
> netcam_keepalive on
73c69
< ffmpeg_bps 44544
---
> ffmpeg_bps 43008
75,76c71,72
< minimum_motion_frames 20
< width 320
---
> netcam_tolerant_check on
> netcam_url http://192.168.1.147:8080/video"
</pre>

chrisf4lc0n, 2017-01-29:
None of the cameras is added as a simple network camera?

mvh0518, 2017-01-29:
I used the 'Add Camera'  'Camera Type': "Network Camera".
See the line:  netcam_url http://192.168.1.147:8080/video"
Are you suggesting I use 'Camera Type': 'Simple MJPEG Camera'?
OK, I'll try that.  
For what it's worth, motion detection once worked on the network camera, but I can't get it back

mvh0518, 2017-01-29:
I tried 'Simple MJPEG Camera', but the comments said "no Motion Detection', so instead
I deleted and recreated the Network Camera  - and IT WORKS.
There are a few parameter differences.  
I save the .conf file and have a diff below of the new one and the saved one.
Full disclosure: At the same time I deleted two dead camera configurations (were cameras 3 and 5)
so this camera moved from camera 4 to camera 3.  If there's any impact because of camera numbers.
Can you tell which parameter below is the change that was necessary?
I see ffmpeg_variable_bitrate, ffmpeg_bps, threshold as the changes.
<pre>
 diff thread-3.conf Save-LRoom.conf
12,13c12
< # @upload_movie on
< # @id 3
---
> # @id 4
16d14
< # @upload_picture on
21c19
< # @name LRoom
---
> # @name Living Room
26a25,26
> # @upload_movie on
> # @upload_picture on
31c31
< threshold 6144
---
> threshold 1843
39,40c39,40
< ffmpeg_variable_bitrate 8192
< ffmpeg_video_codec mp4
---
> ffmpeg_variable_bitrate 20
> ffmpeg_video_codec msmpeg4
42c42
< movie_filename %Y-%m-%d/LRoom%H-%M-%S
---
> movie_filename %Y-%m-%d/LRoom-%H-%M-%S
44c44
< stream_port 8083
---
> stream_port 8084
54c54
< target_dir /var/lib/motioneye/
---
> target_dir /var/lib/motioneye
60,61c60,61
< text_left LRoom
< picture_filename %Y-%m-%d/LRoom%H-%M-%S
---
> text_left Living Room
> picture_filename %Y-%m-%d/LRoom-%H-%M-%S
68a69
> ffmpeg_bps 43008
71a73
>

</pre>

chrisf4lc0n, 2017-01-29:
Good, you got to the bottom of it...
The Simple MJPEG Camera does not support the motion detection at all, hence I was asking if by any chance you had added any of the cameras that way.
I had a similar problem with stalled cameras in the past after updating to the newer version of MotionEye, removing and adding the cameras did the trick.

ccrisan, 2017-08-05:
I'm closing this issue due to lack of activity. Also Simple MJPEG Cameras appear to no longer work, at least on Chrome, due to embedded credentials.

closed 2017-08-05 by ccrisan