motioneye-project/motioneye #431: mjpg-streamer support

jk0, 2017-01-25:
How much work would it be to add support for mjpg-streamer? An example use case would be not needing motion detection, yet still have the ability to stream over the motioneye web interface and record constantly.

My 8 IP cameras are almost melting the hardware running motioneye, which is rather unfortunate since I don't need MD.

ilovelinux, 2019-06-20:
@ccrisan hasn't this feature already been implemented?

jasaw, 2019-06-21:
Enable constant recording and disable motion detection?

Anyway, this question was asked 2 years ago and no update since then. I think this issue can be closed.

ccrisan, 2019-06-21:
I don't think MJPEG Streamer is something to be considered when it comes to motionEye. We've got Motion, a well maintained piece of software that does a far better job than MJPEG Streamer. And it does a lot more than that.

closed 2019-06-21 by ccrisan