motioneye-project/motioneye #432: Wanting to record video and stills to my mac mini's usb connected drive

jat80, 2017-01-26:
How would I go about doing this?

Steps i've done:
1)Enabled remote login on my mac's sharing preferences and confirmed I can ssh into it with the proper login and password.
2) entered details under file storage 
[image]

But it always comes up as "Accessing network share failed unable to mount..."

here are my logs:

Jan 26 16:48:39 raspberrypi meyectl[423]: mount error(5): Input/output error
Jan 26 16:48:39 raspberrypi meyectl[423]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Jan 26 16:48:39 raspberrypi kernel: [19808.913650] CIFS VFS: cifs_mount failed w/return code = -5
Jan 26 16:48:39 raspberrypi meyectl[423]: mount error(22): Invalid argument
Jan 26 16:48:39 raspberrypi meyectl[423]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Jan 26 16:48:39 raspberrypi kernel: [19808.931877] CIFS VFS: cifs_mount failed w/return code = -22
Jan 26 16:48:39 raspberrypi meyectl[423]: mount error(22): Invalid argument
Jan 26 16:48:39 raspberrypi meyectl[423]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Jan 26 16:48:39 raspberrypi kernel: [19808.945819] CIFS VFS: cifs_mount failed w/return code = -22
Jan 26 16:48:39 raspberrypi meyectl[423]: mount error(22): Invalid argument
Jan 26 16:48:39 raspberrypi meyectl[423]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Jan 26 16:48:39 raspberrypi kernel: [19808.960228] CIFS VFS: cifs_mount failed w/return code = -22
Jan 26 16:48:39 raspberrypi meyectl[423]: mount error(5): Input/output error
Jan 26 16:48:39 raspberrypi meyectl[423]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Jan 26 16:48:39 raspberrypi kernel: [19809.046824] CIFS VFS: cifs_mount failed w/return code = -5
Jan 26 16:48:39 raspberrypi kernel: [19809.078249] Status code returned 0xc000006d NT_STATUS_LOGON_FAILURE
Jan 26 16:48:39 raspberrypi kernel: [19809.078280] CIFS VFS: Send error in SessSetup = -13
Jan 26 16:48:39 raspberrypi kernel: [19809.078541] CIFS VFS: cifs_mount failed w/return code = -13
Jan 26 16:48:39 raspberrypi meyectl[423]: mount error(13): Permission denied
Jan 26 16:48:39 raspberrypi meyectl[423]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Jan 26 16:48:39 raspberrypi meyectl[423]: ERROR: failed to mount smb share "//192.168.1.220/"Mac Mini"" at "/media/motioneye_192_168_1_220__mac_mini__####"
Jan 26 16:48:39 raspberrypi meyectl[423]: ERROR: access to network share //192.168.1.220/"Mac Mini" failed: cannot mount network share
Jan 26 16:49:00 raspberrypi kernel: [19829.795888] w1_master_driver w1_bus_master1: Family 0 for 00.998000000000.a2 is not registered.

ccrisan, 2017-01-26:
Why do you use quotation marks all over the place? See [this](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#network-shares).

jat80, 2017-01-26:
My network path has spaces in the name so I thought I had to use quotation marks or the pi won't recognise it?

ccrisan, 2017-01-26:
Try using spaces - it should automatically be quoted. If it doesn't work, try a share without spaces. Also the `/Volumes/...` path looks suspicious. It should be something like `/cams` or `/motionEye`.

jat80, 2017-01-26:
The full path is this:

/Volumes/500GB Internal/motionEye/videos

ccrisan, 2017-01-26:
Make sure it works on a windows machine, just like indicated [here](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#network-shares). When it does, identify the path parts and use them to configure the sharing in motionEye.

jat80, 2017-01-26:
I don't use windows. It's a mac as mentioned in the title.

chrisf4lc0n, 2017-01-26:
Try single quotation marks instead of doubles...

chrisf4lc0n, 2017-01-26:
Or /Volumes/500GB\ Internal/motionEye/videos

jat80, 2017-01-27:
None of the above works.  Thanks for trying to help anyway.  If anyone else has got it working on a mac  please let me know how you did it.  Seems like the software to use network shares is not setup for mac, only windows and linux.

chrisf4lc0n, 2017-01-27:
Samba/CIFS works nearly identical on all Unix based OSes, regardless if it
is Mac OS, Ubuntu or FreeBSD, so I cannot see the reason why the same share
works somwhere else, when on MotionEye it doesn't.
Have you got the share mapped anywhere else and working?
Can you change the name of the share to something without​ the spaces?

jat80, 2017-01-28:
changed to my local /Users folder and still no luck.  I've tried every single combination under the sun.  If anyone actually has it working with a mac please let me know.  

yes I can understand it works regardless of OS but would like to hear from someone using a mac network share to chime in.

chrisf4lc0n, 2017-01-28:
Can you confirm you can access the mounts from any other computer?

ccrisan, 2017-01-29:
@jat80 `/Users...` sounds like an absolute path on your Mac. This can't be your share name, unless you have shared your entire system root (`/`). What have you shared on your Mac? How do your shares look like?

I don't have experience with Mac at all but it can't be that different from other OSes.

chrisf4lc0n, 2017-01-29:
I am still putting my bet on something being wrong on the Mac side.
@ccrisan Mac OS CIFS config is identical to any other Linux/Unix based OS if you do it from the command line...

jat80, 2017-01-31:
got it working.

I used a old NAS i had with standard network shares on that and confirmed it worked.  Tried recording to it and taking stills via motion detection and all ok.

Back to the mac - I changed it back to the mac details but this time I mapped a single folder (in the root directory of the drive) rather than drive.  it said it mapped successfully but then the camera would not display for some reason.

Re-booted the pi.  changed storage location back to local then back to network mac folder again and it seems to be working ok!

Thanks for your help.

Just out of interest what is the best video format to record in?  I tried mp4 which it defaulted to but when downloading the video files it would have the file name.mp4.mpg??  And it wouldn't play.  Switched it to mkv and it works ok.

Thanks again everyone - and sorry for being such a noob!  lol

chrisf4lc0n, 2017-02-01:
It is your browser changing the extension, while saving the file not MotionEye. Since it is most probably not recognising the *.mkv it leaves it alone, but it tries to be clever with mp4 format.

jat80, 2017-02-02:
thanks @chrisf4lc0n.

When the file name.mp4.mpg downloaded I changed it to file name.mp4 but it still wouldn't play and gave an error message.  It would be more useful in mp4 as then can play easily on apple products.  Is it the browser that is causing the corruption when changing the file type?  I use safari.

closed 2017-04-03 by jat80