motioneye-project/motioneye #433: Problem adding Remote motionEye Camera

stevedee, 2017-01-28:
I have motionEye running on a Pi-0 Raspbian Jessie OS.

On a remote laptop I'm also running motionEye and I can "Add Camera" using "Camera Type: Simple MJPEG Camera, URL: http://{Pi ip address}:8081". That works fine.

But if I "Add Camera" using "Remote motionEye Camera, URL: http://{Pi ip}:8765" I just get a spinning icon (i.e. no video). It seems to pick up the correct Camera Name and Camera Device changes to: http://{Pi ip}/config/1

I'm probably missing something really obvious. Anyone know what?

ccrisan, 2017-01-28:
Make sure you run the same motionEye version on both devices. From what you describe, everything should work as expected. Can I see some logs (from both devices)?

chrisf4lc0n, 2017-01-28:
I am finding creating a stream with [https://github.com/revmischa/rtsp-server](https://github.com/revmischa/rtsp-server) to be much less resource hungry and more stable on the occasions.

chrisf4lc0n, 2017-01-28:
Using [https://github.com/mpromonet/v4l2rtspserver](https://github.com/mpromonet/v4l2rtspserver) has proven to be less resource hungry and on some occasions more stable.

stevedee, 2017-01-29:
Thanks for the quick response. I'm still feeling my way around motionEye, so please be gentle with me.

The motionEye version on both the Pi and the remote laptop (running Mate OS) is 0.35.1.

I'm using /home/{user}/.motion for file storage, but I don't see any motioneye.log files. On the Pi I have a tasks.pickle file which is updated when motionEye starts, but contents don't mean much to me. At least this proves that the .motion folder is write-able.

The "log_level" is currently set to "info" so do I need to change this, and what is the catch-all messages level?

Probably unrelated, but I notice motion control port 7999 does not respond (but 8080 does).

Thanks once again.

chrisf4lc0n, 2017-01-29:
The motioneye.log should be in /var/log/ and change the log level to debug before posting.

ccrisan, 2017-01-29:
@stevedee, changes are that you're on systemd. If that is the case, you can obtain the log with `journalctl -u motioneye > motioneye.log`.

And as @chrisf4lc0n indicated, when debugging, you should set the `log_level` to `debug`.

Regarding the port 7999, it should *not* be accessible from outside of your machine. That's the whole point, for it to be controllable from within (i.e. by the local motionEye process), but not from the outside.

stevedee, 2017-01-30:
OK, have skimmed notes on systemd so now have better idea what is does, and understand that logs redirect in motioneye.conf probably does not work with systemd.

Set log level to Debug. Started Pi and laptop, and opened browser on laptop with "add camera" already configured for Remote motionEye Camera.

Did: journalctl -u motioneye > MateMotionEye on Mate laptop:-

[MateMotionEye.txt](https://github.com/ccrisan/motioneye/files/739052/MateMotionEye.txt)

...and: sudo journalctl -u motioneye > PiZeroMotionEye via VNC to Pi-Zero
... which returned an 'unable to resolve host' type message, but sill returned the log file:-

[PiZeroMotionEye.txt](https://github.com/ccrisan/motioneye/files/739055/PiZeroMotionEye.txt)

Thanks once again for taking the time to look at this.

MichaIng, 2026-10-12:
I stumbled over here while searching for something entirely different. @stevedee I checked your (still accessible) debug logs, do not see any issue with the remote motionEye camera on the laptop:
```
Jan 30 13:09:14 BB2017-monitor meyectl[978]:    DEBUG: getting config for camera 4
Jan 30 13:09:14 BB2017-monitor meyectl[978]:    DEBUG: getting config for remote camera 1 on http://192.168.0.25:8765/config/1
Jan 30 13:09:14 BB2017-monitor meyectl[978]:    DEBUG: 200 GET /config/4/get/?force=true&_=XXXX&_username=admin&_signature=XXXX(192.168.0.10) 143.48ms
```
If the video image on the web UI is blank, it might be a bandwidth issue, too.

But the issue is so old, that I think it can be closed, unless you still face the same with motionEye 0.44.0.

closed 2026-10-12 by MichaIng