motioneye-project/motioneye #434: Cannot find "Expert Settings"

Budman1758, 2017-01-30:
Title says most of it.  Expert settings are not anywhere I can find.  I am logged in as admin and "advanced settings" enabled.  This is true on both the camera page and the hub page.  I have tried with 3 different browsers too.  Any ideas?

spinoinside, 2017-02-25:
Same problem ... Have you solved it ?
My setting:
motionEye Version 0.35.3 
Motion Version 4.0.1 
Raspbian Jessie Lite

ccrisan, 2017-02-25:
Can I see a screenshot of your settings panel?

spinoinside, 2017-02-25:
[
[image]
](url)

ccrisan, 2017-02-25:
You're running motionEye, not motionEyeOS. Expert Settings are a motionEyeOS-specific thing. Any feature you're missing from the Expert Settings, I'm afraid is not available or not covered by plain motionEye.

closed 2017-02-25 by ccrisan

stealthwolf, 2019-12-15:
fuck