motioneye-project/motioneye #435: Play a sound on motion detection

andreaviale, 2017-01-30:
Hi, 
I installed motioneye on Raspbian. I'd like to use the "run a command" function to run a python script to play a sound, I tried everything (changing the directory, adding "python" before the command, etc) but it does not work. 
Any suggestion? 
Thanks

chrisf4lc0n, 2017-01-31:
Where would you like to play sound? Locally or on a remote machine?

andreaviale, 2017-01-31:
Locally, with an external speaker

chrisf4lc0n, 2017-01-31:
Do:
sudo apt-get install mpg321
then in the Run Command section add:
mpg321 with a full path to the file you want to play.
For more info do mpg321 -h

andreaviale, 2017-02-01:
Thanks for the suggestion but it does not work with mpg321.
However I've just found a solution. Placing the sound file in /usr/share/sounds/alsa it does work both with omxplayer or aplay.

acujl, 2018-03-25:
sudo dosen't exist, apt-get dosen't exist, how I can install something or make something work in motioneye??

izfari, 2018-06-20:
Hi adreaviale,

I'm just install motioneye on my rasp. On your solution above, your path is /usr/share/sounds/alsa. However, when I inspect my directory path, I dont have Sounds directory. Instead, after folder share, there is folder alsa.

So my question is, should I create new folder, named it Sounds can move folder alsa into it. Or, I just placed my sound file in folder alsa, without folder Sounds

ilovelinux, 2019-06-20:
Did you check that audio device is not set to mute and volume is not set to 0? Did you try to play a sound with aplay? It's not a MotionEye issue since the command run but the system can't play the sound. Try the command via ssh first

ccrisan, 2019-06-21:
This is not a valid motionEye issue. If you want to tweak stuff and you're looking for help, please use our forum: https://groups.google.com/forum/#!forum/motioneye

closed 2019-06-21 by ccrisan