motioneye-project/motioneye #436: Max File Upload Size (128MB). Possible to increase?

apastel, 2017-02-02:
After noticing that some recorded video captures were not being uploaded using Gdrive, I peeked in the logs and saw several errors about files exceeding the maximum file size for upload: 128MB.

I noticed this is not configurable and is hard coded into `uploadservices.py`
```
MAX_FILE_SIZE = 128 * 1024 * 1024 # 128 MB
```

Is there any reason this max file size should not be increased? Is it possible for a user like me who is using one of the motionEye SD card images to increase this limit, or would I need to edit the source and build from scratch? I missed some very juicy videos due to this limit while on vacation and I'd like to make sure that doesn't happen again.

chrisf4lc0n, 2017-02-02:
By all means you can change it to whatever you want, unfortunately when Calin releases new version of MotionEye it will be overwritten to whatever he sets at the time.

apastel, 2017-02-02:
Thanks.
I'm using the MotionEye OS so I guess I wouldn't be able to do that. Is there any way for me to change it on my MotionEye OS system?

chrisf4lc0n, 2017-02-02:
Just edit uploadservices.py on your MotionEye Os and change the value to what you need.

apastel, 2017-02-03:
Ok.
But there isn't an `uploadservices.py` on my MotionEye OS. Just a compiled bytecode file called `uploadservices.pyc`.cat 

```
root@meye-0103e1de /]# find . -name uploadservices*
./usr/lib/python2.7/site-packages/motioneye/uploadservices.pyc
./data/etc/uploadservices.json
```

That's why I was wondering if it was possible to adjust, or if I have to edit the source and recompile the OS to make my own image.

chrisf4lc0n, 2017-02-03:
Are you sure that there is no such file as uploadservices.py in /usr/local/lib/python2.7/dist-packages/motioneye?
I could be wrong, as I have never used the MotionEye OS just MotionEye on Jessie/Ubuntu.

apastel, 2017-02-03:
Yes, I'm sure. The filesystem structure is different on MotionEye OS. There is no `/usr/local` or `dist-packages` directory. The closest directory to the one you mentioned is the `/usr/lib/python2.7/site-packages/motioneye` directory, which contains the `uploadservices.pyc` file I mentioned, along with a bunch of other non-human-readable`.pyc` bytecode files. The only `.py` files on my entire filesystem are:

```
./usr/bin/pilprint.py
./usr/bin/pilfont.py
./usr/bin/pildriver.py
./usr/bin/pilconvert.py
./usr/bin/event_rpcgen.py
./usr/bin/raspimjpeg.py
./usr/bin/pilfile.py
./usr/lib/libstdc++.so.6.0.20-gdb.py
```
Perhaps I can figure out a way to decompile the `uploadservices.pyc` file into a readable state, edit it, and recompile it.

I'd like to suggest making the max file size a configurable option in the MotionEye user interface for future builds, if possible.

chrisf4lc0n, 2017-02-03:
You can always branch the git out, make the changes and update it from
there.

apastel, 2017-02-03:
Ok.
I should also probably be filing this issue under the `motioneyeos` repository, instead of `motioneye`.

ccrisan, 2017-02-03:
Guys, this issue is a valid issue and it belongs with motionEye. I'll make sure to increase that value with in next version.

chrisf4lc0n, 2017-02-03:
I was just editing it in my mobile, but that was rather awkward. Since you
are doing it yourself I shall leave it then.
Thanks.

ccrisan, 2017-02-07:
https://github.com/ccrisan/motioneye/commit/07ccb8a1b77893d30cf655f241f08de5cf443484 increases MAX_FILE_SIZE to 1GB

closed 2017-02-07 by ccrisan