motioneye-project/motioneye #437: mask creation and rotation of picture is inconsistent

hansrijn, 2017-02-03:
When I create a mask with edit-mask in the web-front-end with a rotated picture (270 degrees, in my case) everything looks fine until the pgm file is saved. Motion log says it tries to resize, but fails.

log lines:
[1:ml1] [WRN] [ALL] get_pgm: The mask file specified is not the same size as image from camera.
[1:ml1] [WRN] [ALL] get_pgm: Attempting to resize mask image from 1280x720 to 720x1280

wrong image:
[image]

when rotated 270 back to 0: gives mirrored pgm picture
[image]

It was my intention to mask everything except for the pathe to my house but this cannot be done with the tool when picture is rotated. As below.
[image]

My work around: no rotation and all is well. I try to find other means to create a mask. Could you  advise?

hansrijn, 2017-02-03:
My way to create a rotated pgm: 
1 download mask_1.pgm to windows7, 
2 convert pgm to jpeg with irfanview, 
3 rotate jpeg with windows7,
4 then convert from jpeg to pgm with irfanview
5 upload to pi's /etc/motioneye

ccrisan, 2017-02-04:
Hmm, the mask should be rotated as well once you have rotated the stream. I'll look into this.

hansrijn, 2017-02-04:
Hi Calin,

That would be nice. For me works the workaround. Probably not many that
need rotation.

I appreciate your program and effort very much,

met vriendelijke groet,

Hans Reijnen
+31 6 3754 5659

ccrisan, 2017-02-07:
@hansrijn can you please describe step by step the problem you're encountering? Ideally you should rotate your camera first, edit your mask and never rotate it back. Doesn't it work that way?

hansrijn, 2017-02-08:
actions: 1 rotation, 2 edit, save, apply, 3 try to re-edit.

[image]

[image]

[image]

Again, after clear

[image]

[image]

And again
[image]

[image]

hansrijn, 2017-02-08:
I hope this is of help, if you need more, just ask,

regards,

hansrijn, 2017-02-08:
Logging was:

[1:nc1] [ALR] [NET] netcam_handler_loop: Camera handler thread [3] started
[1:ml1] [WRN] [ALL] get_pgm: The mask file specified is not the same size as image from camera.
[1:ml1] [WRN] [ALL] get_pgm: Attempting to resize mask image from 1280x720 to 720x1280

hansrijn, 2017-02-09:
rotation with imagemagick's convert is easier.
apt install imagemagick
convert -rotate 270 original.pgm rotated.pgm

ccrisan, 2017-02-10:
https://github.com/ccrisan/motioneye/commit/43288b22aba1987cea3e24b6576a8395afffeea3 fixes this issue. @hansrijn, thanks again for taking the time to help me reproduce the issue. I wasn't able to reproduce it because I was trying with a MJPEG netcam where the actual resolution was directly detected (and thus the rotation had no negative impact).

closed 2017-02-10 by ccrisan

hansrijn, 2017-02-11:
perfect, works like a charm, thanks.

hansrijn, 2017-02-11:
Ai, yes editing was like a charm, but....
the mask seems not to be used. The debug output says it is, yet the actual detection uses the entire picture. And, what is more, it now detects motion where no motion is. I get almost continuous saved pictures, where the "frame change window" shows the entire picture.

[image]

[image]

All pictures in above list are identical (but overlay text).

from syslog, maybe of help,

Feb 11 15:12:43 harijnmotion motion: [1:ml1] [DBG] [EVT] exec_command: Executing external command '/usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop 1'
Feb 11 15:12:43 harijnmotion motion: [1:ml1] [NTC] [ALL] motion_loop: End of event 15
Feb 11 15:12:43 harijnmotion meyectl[413]: DEBUG: recevied relayed event stop for thread id 1 (camera id 1)
Feb 11 15:12:43 harijnmotion meyectl[413]: DEBUG: clearing motion detected for camera with id 1
Feb 11 15:12:43 harijnmotion meyectl[413]: DEBUG: 200 POST /_relay_event/?_username=admin&event=stop&thread_id=1&_signature=26ad2a9135dd82501c7e1b4da2e6679aa8deb3f6 (127.0.0.1) 5.90ms

If you need more, pls ask.