motioneye-project/motioneye #438: Nginx configuration : Request some help from community

samfin13, 2017-02-07:
I use Motioneye since one year and today I try to use with a Let's encrypt certificate but without success.

My configuration :
Raspbian 8.0
Motioneye 0.35.1
Motionversion Unofficial-Git-UNKNOWN
nginx version: 1.6.2

I tried to help me from this page :
- https://github.com/ccrisan/motioneye/wiki/Running-Behind-Nginx
- https://homeserver-diy.net/wiki/index.php
- https://github.com/ccrisan/motioneye/issues/298

I created a conf file for nginx `/etc/nginx/sites-available/motioneye.conf` :
```

upstream php-handler {
    server 127.0.0.1:9000;
    #server unix:/var/run/php5-fpm.sock;
}

server {
    listen 8765;
    server_name motioneye.XXXX.ddns.net;
    # enforce https
    return 301 https://$server_name$request_uri;
}

server {
    listen 445 ssl;
    server_name motioneye.XXXX.ddns.net;

    ssl on;
    ssl_protocols TLSv1.2;
    ssl_certificate  /etc/letsencrypt/live/XXXX.ddns.net/fullchain.pem;
    ssl_certificate_key  /etc/letsencrypt/live/XXXX.ddns.net/privkey.pem;
    ssl_dhparam /etc/ssl/private/dh4096.pem;
    ssl_ecdh_curve secp384r1;
    ssl_prefer_server_ciphers on;
    ssl_ciphers EECDH+AESGCM:EDH+AESGCM:AES256+EECDH:AES256+EDH;

    # Path to the root of your installation
    location  /cams/ {
        proxy_pass http://@local_ip_raspberry:8765/;
        proxy_read_timeout 120s;
        access_log off;
    }
}
```

I have uncommented base_path in `/etc/motioneye/motioneye.conf`

```
# the base URL path at which motionEye lives
# change this if you run motionEye behind a reverse proxy (e.g. nginx),
# and you want to make motionEye accessible at a specific base path (e.g. /cams)
base_path /cams
```

In my box, in NAT/PAT rules, I have these redirections :
```
    XXXX.ddns.net:1111 -> @local_ip_raspberry:8735
    XXXX.ddns.net:1112 -> @local_ip_raspberry:445
```
    
Now when I try to access motionEye web page :

- http://XXXX.ddns.net:1111  => only the options like in https://github.com/ccrisan/motioneye/issues/298, it's normal
- https://XXXX.ddns.net:1112 => Nothing :(

If someone more experienced than me see where I made a mistake or what I misunderstood :)

chrisf4lc0n, 2017-02-07:
I have a config for lighttpd if you are interested? No experience with nginx

samfin13, 2017-02-08:
I don't know lighthttpd.
Can it be configure with Let's encrypt certificate ?

chrisf4lc0n, 2017-02-08:
This is exactly what I am doing with the lighttpd.conf below:
```
server.modules = (
	"mod_access",
	"mod_auth",
	"mod_proxy",
	"mod_simple_vhost",
	"mod_alias",
	"mod_cgi",
	"mod_compress",
	"mod_redirect",
        "mod_rewrite",
)

server.document-root        = "/var/www/html"
server.upload-dirs          = ( "/var/cache/lighttpd/uploads" )
server.errorlog             = "/var/log/lighttpd/error.log"
server.pid-file             = "/var/run/lighttpd.pid"
server.username             = "www-data"
server.groupname            = "www-data"
server.port                 = 80

index-file.names            = ( "index.php", "index.html", "index.lighttpd.html" )
url.access-deny             = ( "~", ".inc" )
static-file.exclude-extensions = ( ".php", ".pl", ".fcgi" )

compress.cache-dir          = "/var/cache/lighttpd/compress/"
compress.filetype           = ( "application/javascript", "text/css", "text/html", "text/plain" )

# default listening port for IPv6 falls back to the IPv4 port
## Use ipv6 if available
#include_shell "/usr/share/lighttpd/use-ipv6.pl " + server.port
include_shell "/usr/share/lighttpd/create-mime.assign.pl"
include_shell "/usr/share/lighttpd/include-conf-enabled.pl"

$SERVER["socket"] == "0.0.0.0:5556" {

     proxy.server  = ( "" => 
        (( "host" => "192.168.0.16", "port" => 8765 ))
    )

   ssl.engine = "enable"
    ssl.pemfile = "/etc/lighttpd/lighttpd.pem"

}

```

samfin13, 2017-02-10:
Ok. Thanks.
I'll try this WE.

samfin13, 2017-02-14:
Thanks to chrisf4lc0n for your configuration with lighthttpd but (I don't how :D) I arrived with nginx (with many tets).

I didn't understand the difference between  `/etc/nginx/sites-available` and `/etc/nginx/sites-enabled` … In the first repository, nothing happens :).

I did many tests and one is OK for me.
Here is my nginx conf files, if someone searchs an example `/etc/nginx/sites-enabled/motioneye.conf` :
```
server {
    listen 445 ssl;
    server_name XXX.ddns.net;

    ssl on;
    ssl_protocols TLSv1.2;
    ssl_certificate  /etc/letsencrypt/live/XXX.ddns.net/fullchain.pem;
    ssl_certificate_key  /etc/letsencrypt/live/XXX.ddns.net/privkey.pem;
    ssl_dhparam /etc/ssl/private/dh4096.pem;
    ssl_ecdh_curve secp384r1;
    ssl_prefer_server_ciphers on;
    ssl_ciphers EECDH+AESGCM:EDH+AESGCM:AES256+EECDH:AES256+EDH;

    # Path to the root of your installation
    # root /var/www/cams/;

    location / {
        proxy_pass http://127.0.0.1:8765/;
        proxy_read_timeout 120s;
        access_log off;
    }
}
```
I didn't change the base_path in `/etc/motioneye/motioneye.conf` (I don't understand if I need to create a cams repository and how to access with my ddns address).

closed 2017-02-14 by samfin13