motioneye-project/motioneye #441: Options to disable/enable MotionEye triggers via GPIO

xblox, 2017-02-09:
hi,

as the action buttons featurealarm_on/off_CAMERA_ID_on/off seems to have no impact, I am looking now for options to disable/enable motion-detection triggers (cmd/webhook/mail).

My script to read a GPIO is fairly simple and works so far, consumable by a bash script: 

```python
import RPi.GPIO as GPIO
import sys
GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(27, GPIO.IN,pull_up_down = GPIO.PUD_DOWN)
GPIO.setup(17, GPIO.OUT)

GPIO.output(17, True)
GPIO.input(27)
if(GPIO.input(27) == False):
    exit('button is off')
elif (GPIO.input(27) == True):
    exit('button is on')
```

```bash
#!/usr/bin/env bash
result=$(python /testbutton.py)
if [ "$result" == "button is on" ]; then
    echo "button is on"
    return TRUE
fi
if [ "$result" == "button is off" ]; then
    echo "button is off"
    return FALSE
fi
```

However, is there any chance to silence the motion detection triggers without forking motioneye ?

I can image that you simply extend your system for one more setting, a field to a bash script which returns true/false. Using the result to disable "cmd/webhook/mail" should work for lots of people asking basically the same thing. 

any thoughts welcome, I am ok with a fork, its just annoying to maintain.

greets,
g

xblox, 2017-02-09:
references : 
- https://github.com/ccrisan/motioneye/issues/203
- https://github.com/ccrisan/motioneyeos/issues/80

and I think there is at least a dozen more...

xblox, 2017-02-09:
changing relay_event.sh for that hook might be fully enough, though

xblox, 2017-02-10:
indeed,

1. create a python file /testbutton.py: 

```python
import RPi.GPIO as GPIO
import sys
GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(27, GPIO.IN,pull_up_down = GPIO.PUD_DOWN)
GPIO.setup(17, GPIO.OUT)

GPIO.output(17, True)
GPIO.input(27)
if(GPIO.input(27) == False):
    print('no')
elif (GPIO.input(27) == True):
    print('yes')

```

2. change /usr/local/lib/python2.7/dist-packages/motioneye-0.35.1-py2.7.egg/motioneye/scripts/relayevent.shrelayevent.sh : 

after filename="$4" insert : 

```sh
result=$(python /testbutton.py)
if  test "$result" = "no" ; then
    exit
fi
```

done, extending it per camera should be a piece of cake

closed 2017-02-10 by xblox

chrisf4lc0n, 2017-02-12:
Looks good to me... Branch it and I am sure that Calin will merge it if it works.
It would be very useful if it was possible to make it to run on a separate Pi with buttons to enable/disable MotionEye functions.

bend94, 2018-09-21:
Hi
Can you help me please, i modified that file relayevent.sh
but still recording images , please advise me
regards

Here is my code

#!/bin/bash

if [ -z "$3" ]; then
    echo "Usage: $0 <motioneye.conf> <event> <thread_id> [filename]"
    exit -1
fi

timeout=5
 #echo "$1 $2 $3 "`date`>>/ramdisk/motion.log
motioneye_conf=$1
if [ -f "$motioneye_conf" ]; then
    port=$(cat $motioneye_conf | grep -E '^port' | cut -d ' ' -f 2)
    conf_path=$(cat $motioneye_conf | grep -E '^conf_path' | cut -d ' ' -f 2)
    if [ -n "$conf_path" ]; then
        motion_conf="$conf_path/motion.conf"
        if [ -r "$motion_conf" ]; then
            username=$(cat $motion_conf | grep 'admin_username' | cut -d ' ' -f 3)
            password=$(cat $motion_conf | grep 'admin_password' | cut -d ' ' -f 3 | sed -r 's/[^][a-zA-Z0-9/?_.=&{}":, _]/-/g')
        fi
    fi
fi

test -z "$port" && port="8765"
test -z "$username" && username="admin"

event="$2"
thread_id="$3"
filename="$4"
alarme=`curl -s http://192.168.0.4:8888/alarme`
  # echo  "Alarme vaut "$alarme>>/ramdisk/motion.log
if [[ ${3} == "2" && $alarme == "false" && ${2} == "picture_save" ]]; then
        echo "Alarme OFF , Camera 2 , picture_save JE N ENREGISTRE PAS "$1" "$2" "$3" "$alarme>>/ramdisk/motion.log
        exit
fi
echo "Passage dans uri" >>/ramdisk/motion.log
uri="/_relay_event/?_username=$username&event=$event&thread_id=$thread_id"
data="{\"filename\": \"$filename\"}"
signature=$(echo -n "POST:$uri:$data:$password" | sha1sum | cut -d ' ' -f 1)

curl -s -S -m $timeout -H "Content-Type: application/json" -X POST "http://127.0.0.1:$port$uri&_signature=$signature" -d "$data" >/dev/null