motioneye-project/motioneye #442: Picture misformed // what are best options to reduce cpu

hansrijn, 2017-02-11:
Occasionally I get a pic as below, which is saved after a motion detection. Is there anyone that has seen this before and knows what causes this?
I suspect a cpu overload cannot yet confirm that. My cam has 1280*720 resolution, gives 2 frames/s, sends rtsp stream and motioneye stores on usb-stick and google drive when motion detected (mask used).  

normal load, no motion detected, on pi3:

CPU MEM Program
16.6  2.2 /usr/bin/python /usr/local/bin/meyectl
78.9  5.9 /usr/bin/motion -n
Uptime output:
 13:29:08 up 1 day, 16 min,  1 user,  load average: 1.35, 1.16, 1.08

[image]

What else could I best do to reduce cpu load?

hansrijn, 2017-02-13:
yep, can confirm it is a matter of to high cpu usage. Tried to run unrotated, which saves about 20% of cpu. But this was not enough. Then I decided to run without mask en that seems ok, I get the disformed pics 3 times in last 4 hours. Nevertheless, I would of course run with mask (or rotate the cam). Must I wait for the pi4 or has anyone an idea of cpu reduction without to much lost of functionality?

hansrijn, 2017-02-13:
just saw that motion recognised this isue already, "Excessive CPU usage for RTSP"

JamBalaya56562, 2026-11-07:
Closing as answered, as you confirmed that the misformed pictures came from too high CPU load on the Pi 3. Lowering the resolution or frame rate, or leaving out rotation and masks as you tried, reduces that load.

closed 2026-11-07 by JamBalaya56562