motioneye-project/motioneye #445: Delay between email notifications

psi3000, 2017-02-15:
I would like to have a 30 minute delay between email notification.  I have tried creating a program under data called delay.py.  I have never programmed at all. but here is what i came up with:

main
{
delay (1800000)
}

I then have "data/delay.py" under the run a command.
this does not work.
Any help would be appreciated.

thanks,

chrisf4lc0n, 2017-02-16:
Instead of using the integrated send email function. You could create email sending script with 30 minutes sleep and run it under "Run Command" after "Motion Detection".

psi3000, 2017-02-16:
Ill have to Google a lot of this.  However i have setup Nagios using smtp and mailutils.  Isnt mpack the same thing?  Ill  also have to look up how to create a PID file.  You dont by chance have a step by step link you could provide for someone that has no idea how to code do you?
Im guessing the example script you provided would be the delay.py file and I would run this in the "run a command" part of motioneye?
thanks for the reply!

chrisf4lc0n, 2017-02-16:
I would probably use the flock instead.
What I am finding difficult to attach let's say 5 consecutive image files created by MotionEye.

1. Install ssmtp and mpack.
2. Configure ssmtp.
3. Write a script sending emails (as above). Make it executable.
4. Execute it in the "Run Command" field MotionEye.

The script I provided is a simple bash script, I do not know Python.
mpack allows you to attach files to the emails.

JamBalaya56562, 2026-11-09:
Limiting how often notifications are sent is tracked in #2379, which also has the discussion on how it could work. Closing this one as a duplicate; the script workaround above still works meanwhile.

closed 2026-11-09 by JamBalaya56562