motioneye-project/motioneye #446: MQTT (or other) Support?

stickystyle, 2017-02-16:
Have you given any thought to publishing events on a message bus, such as MQTT? 
For me it would be beneficial to have events or paths to new videos published for easy integration into my home automation system, issue https://github.com/ccrisan/motioneye/issues/132 could maybe be resolved by just sending events onto the bus and let the consumers do what they want with them rather than local scripts that get executed.

entmike, 2017-03-12:
+1 for this.  That would be nice to have it part of the image.  I was just thinking about installing mosquitto and wrapping mine in a script, too.  Sounds like that's the current approach to do then?

thundergreen, 2017-03-14:
+1 I totally agree. we would like having something to implement this awesome cam szuff in Home-Assistant :)

thundergreen, 2017-03-14:
@entmike I built an binary sensor for motion detection. You can also use it as binary sensor if files have been created. But nothing more . If you want, join us on github / gitter chat Home-Assistant

JamBalaya56562, 2026-11-15:
motionEye can already hand its events to MQTT with a command: "Run A Command" under Motion Notifications runs when motion starts, and the one under File Storage runs for each saved file with its path as `%f`. For example `mosquitto_pub -h <broker> -t motioneye/camera1/file -m %f` publishes each new file. A built-in MQTT client would be a new dependency, so this is not planned. Adding `mosquitto-clients` to the Docker image is #877.

closed 2026-11-15 by JamBalaya56562