motioneye-project/motioneye #447: Missing Icons on centos 7

romsworld, 2017-02-17:
Hi,

I have installed motioneye on Centos 7. I have some icons not visible. Do you know why ?

[image]

Thanks,

Feb 17 16:53:47 Installed: epel-release-7-9.noarch
Feb 17 16:54:47 Installed: libjpeg-turbo-1.2.90-5.el7.x86_64
Feb 17 16:55:21 Installed: python-backports-1.0-8.el7.x86_64
Feb 17 16:55:21 Installed: python-backports-ssl_match_hostname-3.4.0.2-4.el7.noarch
Feb 17 16:55:22 Installed: python-setuptools-0.9.8-4.el7.noarch
Feb 17 16:55:23 Installed: python2-pip-8.1.2-5.el7.noarch

ccrisan, 2017-02-17:
Make sure you have the file `/etc/mime.types` present. The package on Arch Linux is called `mime-types`.

romsworld, 2017-02-17:
Thanks !
It's provided by mailcap on centos

 yum install mailcap

closed 2017-02-17 by romsworld