motioneye-project/motioneye #448: webdav nexcloud

Philippe-M, 2017-02-21:
Hello, Is it possible to add the ability to mount a webdav space managed by an external nexcloud server to store images and videos?

ccrisan, 2017-02-21:
Is this motionEye or motionEyeOS?

Philippe-M, 2017-02-21:
I'm use motionEyeOs but I do not know if it's up to motionEye or motionEyeOs to handle it

ccrisan, 2017-02-21:
On motionEyeOS you could use FTP, SFTP or SMB to mount your camera system in your nextcloud server. WebDAV is not something that will happen to motionEyeOS.

If you were on Raspbian + motionEye, I'd advise googling for WebDAV Raspbian.

closed 2017-02-21 by ccrisan

WisseHesNL, 2019-02-17:
a good tuturial for mounting webdav on raspbian is https://ajclarkson.co.uk/blog/auto-mount-webdav-raspberry-pi/ it is a bit outdated but it still works