motioneye-project/motioneye #449: MemoryError gdrive uploading file > 300 MB

apastel, 2017-02-21:
After using the new update that removed the 128 MB upload restriction, I get an error now in `/var/log/motioneye.log` when a video capture file greater than roughly 300 MB is attempting to upload using gdrive:

```
2017-02-20 15:42:27: [motioneye]    ERROR: gdrive: request failed: 
2017-02-20 15:42:27: [motioneye]    ERROR: failed to upload file "/data/output/Camera1/2017-02-20/15-38-01.mp4" with service gdrive: 
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 658, in upload_media_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 93, in upload_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 209, in upload_data
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 336, in _request
  File "/usr/lib/python2.7/site-packages/motioneye/utils.py", line 787, in urlopen
  File "/usr/lib/python2.7/urllib2.py", line 154, in urlopen
  File "/usr/lib/python2.7/urllib2.py", line 431, in open
  File "/usr/lib/python2.7/urllib2.py", line 449, in _open
  File "/usr/lib/python2.7/urllib2.py", line 409, in _call_chain
  File "/usr/lib/python2.7/urllib2.py", line 1240, in https_open
  File "/usr/lib/python2.7/urllib2.py", line 1194, in do_open
  File "/usr/lib/python2.7/httplib.py", line 1057, in request
  File "/usr/lib/python2.7/httplib.py", line 1097, in _send_request
  File "/usr/lib/python2.7/httplib.py", line 1053, in endheaders
  File "/usr/lib/python2.7/httplib.py", line 895, in _send_output
MemoryError
```

Files around 250 MB seem to upload ok, but several video capture files have failed to upload and they all have been between 300 - 400 MB. I tried to inspect the python2.7 library (specifically httplib.py) but I didn't see any reason why I should be getting a MemoryError. I am wondering if it's just my hardware (RPi 3 running Raspbian Jessie Lite) or if everyone is having this problem.

ccrisan, 2017-02-21:
I believe you're running out of memory.

apastel, 2017-02-21:
Yes, I am. I was wondering if you were experiencing this issue as well or if it was just a personal problem with my hardware (1 GB RAM), and/or if you had any suggestions, or if it's possible to increase the memory allocated to motioneye / python.

ccrisan, 2017-02-21:
I don't think there's any tweak that will make it work. I don't have media files larger than a few tens of megabytes so I never encountered this issue myself. But it's understandable, given the overhead that python introduces when it comes to memory (or CPU, for that matter), to experience out-of-memory errors with files of hundreds of megabytes.

apastel, 2017-02-21:
Yeah, it must be related to Python because the file uploads successfully if I run `gdrive upload [filename]` outside of motioneye. Might just have to do that whenever there's a MemoryError.

JamBalaya56562, 2026-11-07:
Closing as answered: as ccrisan said above, uploading files of several hundred MB ran out of memory on the 1 GB Pi 3, and `gdrive upload` outside motionEye worked for you. This was with Python 2; if large uploads still fail with a MemoryError on a current release, please open a new issue with the log.

closed 2026-11-07 by JamBalaya56562