motioneye-project/motioneye #451: Can't add RTSP camera due to URL formatting.

hede0n, 2017-02-21:
Cameras I am using are only streaming over RTSP. I cannot add them to motioneye because of different URL formatting they are using: 

rtsp://user:password@**ipaddress**/onvif1

This doesn't seem to be an issue in VLC

I would love to be able to use those camras because they are extremely cheap, very good quality so far, and work well in LAN only (had issues with other cameras)
http://www.banggood.com/Digoo-DG-M1Q-960P-2_8mm-Wireless-Mini-WIFI-Night-Vision-Smart-Home-Security-IP-Camera-Onvif-Monitor-p-1123595.html

ccrisan, 2017-02-21:
Have you tried entering the username & password separately, in their corresponding fields?

hede0n, 2017-02-21:
Yes, I did. When adding new camera after typing IP address forth field with drop down list is showing up correctly so I am assuming Motioneye is recognising camera under this IP. However, picture stays grey forever. 
The only way to see the stream in any software I have tried so far was by using URL formatted like in the issue I have submitted. When using this format in MotionEye it seams to not recognise it...

hede0n, 2017-02-21:
Scrap it! It does work. As RTSP UDP not tcp. Streams takes fair amount of time to show but it does work. Sorry, solved.

ccrisan, 2017-02-21:
Good to hear :)

chrisf4lc0n, 2017-02-21:
On top of that, what I have noticed with some of the RTSP cameras, when you try to add them as TCP and  it does not take it, you have to restart MotionEye to try to add it as UDP otherwise it does not recognise the camera at all.

ccrisan, 2017-02-21:
Anyways if you got it working, I'll close the issue.

closed 2017-02-21 by ccrisan

wouterkabouter77, 2017-03-30:
I have the same camera and problem. How can i add a camera as  UDP camera and what wil be the url?

mikeys4u, 2017-10-22:
After an hour of mucking around in Tinycam monitor for android, the rtsp url is exactly this 
rtsp://user:password@ipaddress/onvif1
using RTSP over UDP (mpeg h264/h265)

Thank you......

chrisf4lc0n, 2017-10-22:
Finding the right streaming URL can be a real pain!  I am glad, that you got to the bottom of it!

mrdmiba, 2019-04-18:
Is user name in rtsp link email address used as login name for Guudgo Cloud or something else?