motioneye-project/motioneye #452: flipping image

asquelt, 2017-02-22:
i have a network camera with image which is flipped (looking at mirror). is it possible to correct it?

ccrisan, 2017-02-23:
I don't think motion allows us to flip the image. [This](http://htmlpreview.github.io/?https://github.com/Motion-Project/motion/blob/master/motion_guide.html) is the most recent version of their doc. If you know a way to make motion flip the image, I'll consider adding the option to motionEye.

asquelt, 2017-05-29:
the upstream has introduced option "flip_axis {v|h|none}". can you update UI so that it allows setting this?

asquelt, 2017-05-29:
@ccrisan, kindly please reopen.
the upstream has introduced option "flip_axis {v|h|none}". can you update UI so that it allows setting this?

ccrisan, 2017-05-30:
Alright. But let's wait for a motion release. Otherwise it'll be impossible to tell whether the binary supports or not these new options.

rjblake, 2018-03-28:
Adding "flip_axis h"  under Video Device->Extra Motion Options flips the Raspberry Pi camera

PegFaceMan, 2022-06-10:
Apologies was being stupid.  Used the existing option for 'Video Rotation' 180

JamBalaya56562, 2026-11-10:
Closing, as Motion's `flip_axis h` (or `v`) under Video Device > Extra Motion Options mirrors the picture, as rjblake confirmed, and Video Rotation covers turning it by 180°. A separate switch for it is not planned.

closed 2026-11-10 by JamBalaya56562