motioneye-project/motioneye #453: Raspberry Pi Spy Cam not working with MotionEye?

caesarx86, 2017-02-23:
Hi,

I'm trying to use Motion Eye with the Raspberry Pi Spy Cam i recently purchased from Adafruit: https://www.adafruit.com/product/1937

It's connected and the tests with raspistill shows it works.
Then I set up MotionEye on Raspbian Jessie and it's accessible in a browser from port 80.

However when i select add camera 
> Camera Type: Local Cam 
> Camera: mmal-1.6 (the only option i get)
When i click OK it sais "An error occured" for an instance and then i get the "no picture"-icon.

Am i forgetting something?

Thanks in advance!

Nick

ccrisan, 2017-02-23:
Please attach your motioneye log.

caesarx86, 2017-02-23:
Hi, where do i find the logfile? Can't find it in /var/log

I completely reinstalled raspbian and now i'm getting 1 step further, it adds the cam but then i get this screen:

[image]

While in RPi_Cam_Web_Interface the cam just works:

[image]

Thanks in advance!

ccrisan, 2017-02-24:
`journalctl -u motioneye > motioneye.log`