motioneye-project/motioneye #455: Cycling the images of the cameras

lozziboy, 2017-02-23:
Connecting motioneye system to a big LCD screen, is useful to have an option that, in monitor view, cycles the videos of the cameras (full screen), one camera per time, and it change the visualized video every 20 seconds.
This is very common in professional security systems.

lozziboy, 2021-06-01:
Please remove the "Stale No Activity 60 Days" label.
This feature is still requested 👍

starbasessd, 2021-06-01:
Feature is enabled in browsers:
https://chrome.google.com/webstore/detail/auto-tab-switch/iagkccjmbfoijhedememehigdjocdkhd?hl=en
Open each camera stream in a new tab
http://motioneye_ip_address:8081
http://motioneye_ip_address:8082
http://motioneye_ip_address:8083
http://motioneye_ip_address:8084
etc.
Then install and run the above add-in (similar add-ins for firefox and other browsers)

starbasessd, 2021-06-01:
The tag is valid. Request is over 4 years old, and even after being tagged, no activity was added even after another 60+ days.