motioneye-project/motioneye #456: Using IP camera aux stream as

tehKaiN, 2017-02-24:
Hi there,

I'm using Armbian running on OrangePi One with two RTSP IP cameras. Since I've installed 2nd camera the CPU usage is very high due to poor OS/HW H.264 support and I'm trying to fight it in various ways.

Would it be possible to add a feature to MotionEye, so image preview & motion detection would take place using lower-resolution secondary stream provided as separate RTSP link, and recording by using main, higher resolution stream? That way, my rig would need to process less pixels to detect general movement by price of higher network usage. Main stream could then be saved directly to file without further processing.

Triggering main stream recording using external cmd is sort of way, but that would mean a time gap between detection and connecting & starting stream dump. Having a process buffering main stream and saving it on trigger with time margins would be a better way to go.

McNobody, 2017-07-07:
I would like see this feature too.

I have Logitech Alert cameras which provides 960x720 video at stream rtsp://xx.xx.xx.xx/HighResolutionVideo and resolution 640x480 at stream rtsp://xx.xx.xx.xx/LowResolutionVideo

It would be awesome to process low resolution to detect motion and then capture snapshots and videos from higher resolution video. The preview at Motioneye desktop could also be lower resolution video. The fullscreen video should propably be with higher resolution.

ektus, 2017-07-26:
This is likely a feature that has to be incorporated in motion, see also https://github.com/Motion-Project/motion/issues/123
Another possibility would be to define two cameras within motioneye, one for each resolution, and start recording on both streams (or only high-res) in case of motion detection on the low-res stream, without activating motion detection for the high-res stream.

tehKaiN, 2018-01-01:
Since https://github.com/Motion-Project/motion/pull/522 has landed to Motion, would it be possible to add such feature for MotionEye?

ccrisan, 2018-01-01:
I'll have to look into it and see how it actually works.

J-C-P, 2024-07-13:
Hello
I don't want to be pushy but...
Is there any news regarding the dual feed management ?
Thanks a lot :-)

zagrim, 2024-07-22:
Oh, it is definitely possible to do that now, but it seems that even now you just need to manually add the hi-res camera URL by adding this to "Extra Motion Options": `netcam_high_url rtsp://xx.xx.xx.xx/HighResolutionVideo` (of course, substitute your actual hi-res camera URL in place of "rtsp://xx.xx.xx.xx/HighResolutionVideo"). And if you have Motion 4.3 or older, use `netcam_highres` instead of `netcam_high_url`.

By adding that, Motion should start saving the hi-res video and doing motion detection on the "normal" camera stream (URL).

If you need to skip all possible processing of the hi-res video stream, you can toggle "Motion passthrough" switch in ME GUI:
[image]

Further details may be found from Motion docs: https://motion-project.github.io/motion_config.html#netcam_high_url

closed 2024-07-22 by zagrim