motioneye-project/motioneye #459: corrupted mp4 files

cmundi, 2017-02-25:
Clean install:
rpi 3 with piCam 2.1
motioneye latest via pip
motion 4.0.1-1
ffmpeg  3.1.1-1 (prebuilt from https://github.com/ccrisan/motioneye/wiki/precompiled/ffmpeg_3.1.1-1_armhf.deb -- not from deb-multimedia)

MotionEye does create mp4 files when motion is detected. But they seem to be corrupted and either don't play at all, play back only one frame (in a 14 MB file!) in VLC on raspbian. I know ffmpeg can be a can of worms. I thought the problem might be with the raspian build of vlc. So I copied the mp4's to a windows box -- same result with VLC and Windows Media Player just crashed.

1. Are the versions above mutually compatible?
2. Where does Motioneye fetch its ffmpeg parameters?

All suggestions welcome. Thanks.

JosephRedfern, 2017-03-03:
I'm experiencing a similar issue. Typically, the first second or so of video comes out smooth, but it quickly degrades and freezes. I have the same setup as you, but am using a USB Webcam.

TobbeJ, 2017-04-02:
i'm also seeing this.
my mp4 files tend to work but is extremely blocky and i bet they have lost frames too.
maybe it cant keep up and is dropping frames during encoding?

is ffmpeg or whatever is encoding the files taking advantage of the hw assisted h264 encoding?
like raspivid or whatever it was

mariocannistra, 2017-04-20:
just downloaded and imaged microsd for R-PI-2.
using picam.
I can download the motion movie with .mp4 extension but cannot play it with any of the players I have on windows 8.1.

TobbeJ, 2017-05-01:
i'm pretty sure i know what the problem is.

when ffmpeg is called it never takes into consideration what the file extension should be of the temporary file and ffmpeg is always feed with -f avi parameter.
so no matter what you pick it will probably confuse ffmpeg a lot and FFMPEG_EXT_MAPPING in the code is never used

so code is probably partially incomplete.

metalshreds, 2017-07-17:
I'm experiencing this exactly on a Rpi3, videos save as mp4, play one second then freeze the rest of the time.

TobbeJ, 2017-07-17:
i solved my problem after a lot of troubleshooting.
turns out the meaning of the quality parameter that gets sent to ffmpeg changed alot between versions.
i think that either during upgrade or possibly wrong defaults in motioneye it got screwed up and this was not obvious at all.

all i had to do was to change the quality slider from 0 or 1 (used to mean autodetect or similar) to much higher, i think i left it near max.

makes sense files are just garbage if quality is set to lowest possible setting.

metalshreds, 2017-07-17:
Is that the setting "Movie Quality" under "Movie"? I have mine set to 75% just bumped it to 100%, I also noticed video memory was set to 128, bumped it to 256. Hopefully that will resolve the problem, thanks!

TobbeJ, 2017-07-17:
i think that's the one, mine was set to 1 which is why i got bad movies.

metalshreds, 2017-07-20:
hmm unfortunately this didn't fix my problem and I set it to 100% quality, also changed the GPU memory to 512mb. recorded movies are very choppy, I'll keep experimenting

timdonovanuk, 2017-09-22:
I'm having this problem on the latest version. The first 4-6 seconds of the movie captured are garbage (at least, all video players just show a black screen). I then get 1 frame displayed and thats it, for the rest of the movie clip.

I've set the camera resolution and frame rate to as low as it will go.

xavrr, 2017-11-14:
Same here on a raspberry pi zero... anyone solved this ?

jasaw, 2017-11-14:
Motion version 4.0.1 has lots of video recording issues. A lot of recording issues have been fixed in motion 4.1. Give it a try:  https://github.com/Motion-Project/motion/releases/tag/release-4.1

kleini, 2017-11-15:
Getting version 4.1 of motion into the Docker images should not be that much work. Will try to get this done this week.

kleini, 2017-11-15:
See https://hub.docker.com/r/ccrisan/motioneye/ for ready to use Docker images containing motion 4.1 and latest motionEye from dev branch. You need to use the images tagged with newMotion-amd64 or newMotion-armhf.

ilovelinux, 2019-06-18:
Issue still here.

danmutblix, 2020-02-21:
You have to play the file in motioneyeos fileplayer, then right-click and save file as.
File should be renderable in VLC

drHuangMHT, 2020-04-29:
> 
> 
> You have to play the file in motioneyeos fileplayer, then right-click and save file as.
> File should be renderable in VLC

I installed motioneye via pip on Raspbian.  But I havn't seen any fileplayer in motioneye webpage. Any details?

danmutblix, 2020-04-29:
@drHuangMHT yes, there are some major differences in the GUI in motioneyeos and packages installed via pip or apt. If you use motioneye standalone os image, then it will work and you will also get full packed features in the GUI. The drawback is that the only OS you will be able to run is motioneye..

danmutblix, 2020-04-29:
@drHuangMHT as an alternative you get rsync or scp your videos folder to your choice of destination

drHuangMHT, 2020-04-29:
> 
> 
> @drHuangMHT as an alternative you get rsync or scp your videos folder to your choice of destination

Thanks! I tried copying files to another folder(by command "cp"), but I still got a black screen. I also tried other codecs, but the results are the same. Must I copy them using rsync or scp?

danmutblix, 2020-04-29:
@drHuangMHT np 👍
F.eks my videos are stored here in /home/pi/Videos, then if I want to copy them to my desktop I just do 
``
NR=$(find /home/pi/Videos -type f | wc -l); sudo rsync -vrltDh --stats /home/pi/Videos/ daniel@10.0.0.124:/home/daniel/desktop/pinoir | pv -leps $NR
``
pv will give you a nice monitoring with a progress bar

danmutblix, 2020-04-29:
I don't remember changing anything in the codecs, I think I use mp4 if I'm not mistaken

drHuangMHT, 2020-04-29:
@danmutblix I can't understand what that command will do.  Why bother? btw I'm new in Linux

JosephRedfern, 2020-04-29:
That command counts the number of files in `/home/pi/Videos`, then kicks of an rsync job to copy files to `/home/daniel/desktop/pinoir` on `10.0.0.124`, The final `| pv -leps $NR` section produces a progress bar, which uses the count (NR, defined at the start of the line) to give you % completion, and works because rsync will kick out a line for every file that's copied.

It's basically a nice way of monitoring rsync progress.

danmutblix, 2020-04-29:
@JosephRedfern explains it nicely

drHuangMHT, 2020-04-29:
@JosephRedfern Thank you very much for your time explaining that command. But why can't I use "cp" to copy the video files? Or just right click then copy and paste? The original file(located in /var/lib/motioneye) also shows a black screen.

danmutblix, 2020-04-29:
@drHuangMHT if you have the fully featured motioneyeOS running as host OS on the PI, yes then you should be able to right click the video, save as .mp4 and use VLC watch the content locally. You have installed motioneye through pip, so you must refer to the docs on how to proceed.
cp does not remote copy, scp and rsync do
You typically use that to copy from your PI server to your client, e.g windows or a GUI based *ux distro

drHuangMHT, 2020-04-29:
@danmutblix Now I get it, but I just want the video to be played properly....... I played the video locally, but I got a black screen. I played it on Windows, but it's also a black screen. I don't know what went wrong, and I want to know how to fix it. I still want my pi to do something else, not just record videos.

danmutblix, 2020-04-29:
@drHuangMHT yes I understand, but as I mentioned, many features in motioneye pip or motioneye from apt seem to be lacking some features as opposed to the full motioneyeOS. If you flash your PI with the motioneyeOS you will understand

drHuangMHT, 2020-04-29:
@danmutblix Fine...... doesn't help but thank you anyway. I'll try the OS.

functionalprototype, 2021-03-13:
I started seeing this after I switched from an Arducam to an official RPi camera.  Note that I didn't reinstall the RPi from scratch, I just switched cameras and loaded the devices for the new camera.

Debugging time!

Raspberry Pi 4 Model B Rev 1.4
Linux version 5.10.17-v7l+ (dom@buildbot) 
motionEye Version | 0.42.1
Motion Version | 4.2.2
macos Firefox 86.0 (64-bit)

What's interesting(?) is that I can download the mp4 file, open it with VLC, but if I try to open it with FF, I get the corrupt message.  Safari won't even let me select the mp4 file for open in the browser.

However, Google Chrome will open the mp4 and play the video.  If I connect to the RPi with Chrome, it will also play the videos as it should.

Did a little more searching on the Internet and it looks like FF not playing H.264 codecs is an issue on some (may?) platforms.  I tried all the other codecs and also tried different resolutions, nothing would play in FF.

Will flash a new image and see if it works if I start from scratch with the new RPi camera.

starbasessd, 2021-03-13:
motioneyeos on the pI4? Or another OS? 
Did you clear cache & cookies on Firefox?

functionalprototype, 2021-03-13:
I just posted the configuration of the RPi running motioneye.  What information is missing?

Yes, I tried again after clearing the cache.  FF won't play the files loaded from disk, entirely bypassing motioneye.

I think something related to installing the RPi camera/drivers is the problem.  Will try and switch back to Arducam before I reflash the sd card.

starbasessd, 2021-03-13:
In the WebGUI it shows as either

OS Version | motionEyeOS 20200606 (Whatever date the image was created)
or
OS Version | Ubuntu 20.04 (Or whatever Base OS it's on
You've just listed the Kernel version, which doesn't help me much...

functionalprototype, 2021-03-13:
motionEye Version | 0.42.1
Motion Version | 4.2.2
OS Version | Raspbian 10

I listed the Linux release version as it has more detail.

This is a headless Rpi4 that I'm accessing from FF on macos, it worked just fine yesterday with the Arducam.

starbasessd, 2021-03-13:
For the kernel, yes. But different OSes act differently, and I try to
reproduce issues here to better help.
Firing up my Mac and an RPi4-4GB/RaspberryPiOS Buster, hold on.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

functionalprototype, 2021-03-13:
I've been developing in C/*nix since SVR0 but am new to the RPi.  If there's any patches you want me to test, I'll pull a tree and do a local build.

starbasessd, 2021-03-13:
Quick & dirty:
OPen MacOS
Install Firefox 86
Open my Camera Hub (port 8765)
Sign in as admin
Click on stream
open Video button
Select an earlier recording from today
Click center of movie, plays fine for multiple cameras and formats,
including H.264/OMX and passthrough.
Format is set to passthrough and works fine.
Can't check on RPi4/PiCam, no ribbon cable. Works fine on PiZeroW with
PiCam.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-13:
I was using FF 86.0.1, and I have the latest VLC also installed.

functionalprototype, 2021-03-13:
Those steps worked perfectly with Arducam.  I suspect it's something wonky in driverland.  I'll pull/build a tree and see what I can find out.

starbasessd, 2021-03-13:
Driver for PiCam are built into the kernel (CSI driver)
Once the camera is recognized, it's up to motion to handle the video.
How did you install? Which set of instructions?
Did you try to do the 4.3.2 build of motion? Do you build it from source?
THere are issues doing so, especially with OMX.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

functionalprototype, 2021-03-14:
I installed using these instructions:  https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
which gave me 4.2.2.  Going to do a build of 4.3.2 on a fresh sd card and see what happens.

functionalprototype, 2021-03-14:
Ok, so here's some good news.

I flashed a new SD card with the latest Raspberry 
I mostly followed the install instructions in the previous comment -- I did: sudo apt-get update && sudo apt-get upgrade just because
To set up the camera I did: sudo cp camera4-dist.conf camera-1.conf
In the web UI for adding a new camera, I picked the camera type of MMAL Camera
Using the web interface I moved the resolution to 1920x1080
Then I set the camera to record on motion and things worked as expected.

So, for MMAL cameras, I think installing the upgrades for the RPi camera adapter on an existing system might be problematic.

starbasessd, 2021-03-14:
On a fresh install, following the instructions posted, (used a PiZeroW and
RaspberryPiOS 20210111 and enabling PiCam in raspi-config) it
auto-discovers MMAL cameras (CSI Cable) automagically, no need to
copy/move  files around.
So this leads me to suspect you may be using a MIPI Arducam, or one with
the enhanced features/chipsets, rather than the Native Support Arducam?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

functionalprototype, 2021-03-14:
I had the same experience with a MMAL Arducam on several RPi 3 and RPi 4.  It was switching to the RPi camera that caused the problems.  I had to reinstall to get that to work.

JamBalaya56562, 2026-11-07:
Closing, as this was about Motion 4.0.1, which had many recording problems that Motion 4.1 fixed, as jasaw wrote above. Please open a new issue with the Motion and motionEye versions if movies still come out broken with a current release.

closed 2026-11-07 by JamBalaya56562