motioneye-project/motioneye #460: Time-lag

cmundi, 2017-02-25:
rpi3 + piCam2

Is it normal to consistently have a four-second delay (approximate) between changes in the real world and what we see in the feed on screen? 

Running the piCam with python test scripts shows virtually no lag between waving my hand and seeing it on screen.  With motioneye there is about a 4-second lag. MotionEye definitely keeps up. It's not falling further behind, It starts out four seconds late and holds that delay.  It this just a queuing artifact of setting up the v4l -> motion -> motioneye pipeline?

Thanks!

ccrisan, 2017-02-26:
The delay is normal and is due to the processing time.

closed 2017-02-26 by ccrisan

cmundi, 2017-02-26:
Thanks for confirming.