motioneye-project/motioneye #461: Problem with Jessie/RaspiCam

Dodutils, 2017-02-26:
Hello,

I installed motionEye on a Jessie/Raspi2/RaspiCam V1

Raspistill test OK

added /etc/modprobe => bcm2835-v4l2

/dev/video0 present

but get grey windows "unable to open video device"

If I try to start motion itself from /etc/motioneye

[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [WRN] [ALL] thread config option deprecated use camera
[0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1 Started
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1:ml1] [ERR] [ALL] motion_init: Error capturing first image
[1:ml1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1:ml1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1:ml1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera

if I start motion from outside /etc/motioneye I get also errors : 

[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motion/motion.conf
[0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1 Started
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
[0:motion] [NTC] [ENC] ffmpeg_init: ffmpeg libavcodec version 56.1.0 libavformat version 56.1.0
[0:motion] [NTC] [ALL] main: Camera 0 is from /etc/motion/motion.conf
[0:motion] [NTC] [ALL] main: Camera 0 is device: /dev/video0 input -1
[0:motion] [NTC] [ALL] main: Stream port 8081
[0:motion] [NTC] [ALL] main: Waiting for threads to finish, pid: 8264
[0:web_control] [NTC] [STR] http_bindsock: error binding on 127.0.0.1 port 8080
[0:web_control] [NTC] [STR] motion_web_control: motion-httpd thread exit
[1:ml1] [NTC] [ALL] motion_init: Camera 0 started: motion detection Enabled
[1:ml1] [NTC] [VID] vid_v4lx_start: Using videodevice /dev/video0 and input -1
[1:ml1] [NTC] [VID] v4l2_get_capability:
------------------------
cap.driver: "raspicam"
cap.card: "Camera Board OV5647"
cap.bus_info: "CSI"
cap.capabilities=0x01000001
------------------------
[1:ml1] [NTC] [VID] v4l2_get_capability: - VIDEO_CAPTURE
[1:ml1] [NTC] [VID] v4l2_get_capability: - READWRITE
[1:ml1] [NTC] [VID] v4l2_select_input: name = "default", type 0x00000002, status 00000000
[1:ml1] [NTC] [VID] v4l2_select_input: - CAMERA
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1:ml1] [NTC] [VID] v4l2_do_set_pix_format: Testing palette YU12 (320x240)
[1:ml1] [NTC] [VID] v4l2_do_set_pix_format: Using palette YU12 (320x240) bytesperlines 480 sizeimage 115200 colorspace 00000000
[1:ml1] [NTC] [VID] v4l2_scan_controls: found control 0x00980900, "brightness", range 0,100
[1:ml1] [NTC] [VID] v4l2_scan_controls:         "brightness", default 50, current 50
[1:ml1] [NTC] [VID] v4l2_scan_controls: found control 0x00980901, "contrast", range -100,100
[1:ml1] [NTC] [VID] v4l2_scan_controls:         "contrast", default 0, current 0
[1:ml1] [NTC] [VID] v4l2_scan_controls: found control 0x00980902, "saturation", range -100,100
[1:ml1] [NTC] [VID] v4l2_scan_controls:         "saturation", default 0, current 0
[1:ml1] [NTC] [VID] v4l2_scan_controls: found control 0x0098090e, "red balance", range 0,800
[1:ml1] [NTC] [VID] v4l2_scan_controls:         "red balance", default 100, current 100
[1:ml1] [NTC] [VID] v4l2_scan_controls: found control 0x0098090f, "blue balance", range 0,800
[1:ml1] [NTC] [VID] v4l2_scan_controls:         "blue balance", default 100, current 100
[1:ml1] [NTC] [VID] v4l2_scan_controls: found control 0x08000000, "shutter speed", range 0,65535
[1:ml1] [NTC] [VID] v4l2_scan_controls:         "shutter speed", default 0, current 0
[1:ml1] [NTC] [VID] v4l2_scan_controls: found control 0x08000001, "zoom factor", range 1,8
[1:ml1] [NTC] [VID] v4l2_scan_controls:         "zoom factor", default 0, current 1
[1:ml1] [NTC] [VID] vid_v4lx_start: Using V4L1
[1:ml1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[1:ml1] [ERR] [ALL] motion_init: Error capturing first image
[1:ml1] [NTC] [STR] http_bindsock: listening on 127.0.0.1 port 8081
[1:ml1] [NTC] [ALL] motion_init: Started motion-stream server on port 8081 (auth Disabled)
[1:ml1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[1:ml1] [NTC] [VID] vid_close: Closing video device /dev/video0
[1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera

Any idea ?

regards.

ccrisan, 2017-02-26:
`error binding on 127.0.0.1 port 8080` - make sure you're not running something else (e.g. another motion instance) on that port.

Dodutils, 2017-02-26:
Yes but anyway it says it cannot capture from camera whenever biding is good or not.

But here is a log with no binding errorr and still do not work :

```
------------------------
cap.driver: "raspicam"
cap.card: "Camera Board OV5647"
cap.bus_info: "CSI"
cap.capabilities=0x01000001
------------------------
[1:ml1] [NTC] [VID] v4l2_get_capability: - VIDEO_CAPTURE
[1:ml1] [NTC] [VID] v4l2_get_capability: - READWRITE
[1:ml1] [NTC] [VID] v4l2_select_input: name = "default", type 0x00000002, status 00000000
[1:ml1] [NTC] [VID] v4l2_select_input: - CAMERA
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1:ml1] [NTC] [VID] v4l2_do_set_pix_format: Testing palette YU12 (320x200)
[1:ml1] [NTC] [VID] v4l2_do_set_pix_format: Using palette YU12 (320x200) bytesperlines 480 sizeimage 99840 colorspace 00000000
[1:ml1] [NTC] [VID] v4l2_scan_controls: found control 0x00980900, "brightness", range 0,100
[1:ml1] [NTC] [VID] v4l2_scan_controls:         "brightness", default 50, current 50
[1:ml1] [NTC] [VID] v4l2_scan_controls: found control 0x00980901, "contrast", range -100,100
[1:ml1] [NTC] [VID] v4l2_scan_controls:         "contrast", default 0, current 0
[1:ml1] [NTC] [VID] v4l2_scan_controls: found control 0x00980902, "saturation", range -100,100
[1:ml1] [NTC] [VID] v4l2_scan_controls:         "saturation", default 0, current 0
[1:ml1] [NTC] [VID] v4l2_scan_controls: found control 0x0098090e, "red balance", range 0,800
[1:ml1] [NTC] [VID] v4l2_scan_controls:         "red balance", default 100, current 100
[1:ml1] [NTC] [VID] v4l2_scan_controls: found control 0x0098090f, "blue balance", range 0,800
[1:ml1] [NTC] [VID] v4l2_scan_controls:         "blue balance", default 100, current 100
[1:ml1] [NTC] [VID] v4l2_scan_controls: found control 0x08000000, "shutter speed", range 0,65535
[1:ml1] [NTC] [VID] v4l2_scan_controls:         "shutter speed", default 0, current 0
[1:ml1] [NTC] [VID] v4l2_scan_controls: found control 0x08000001, "zoom factor", range 1,8
[1:ml1] [NTC] [VID] v4l2_scan_controls:         "zoom factor", default 0, current 1
[1:ml1] [NTC] [VID] vid_v4lx_start: Using V4L1
[1:ml1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[1:ml1] [ERR] [ALL] motion_init: Error capturing first image
[1:ml1] [NTC] [STR] http_bindsock: listening on any IPv4 address port 8081
[1:ml1] [NTC] [ALL] motion_init: Started motion-stream server on port 8081 (auth Disabled)
[1:ml1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 21 items
[1:ml1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[1:ml1] [NTC] [VID] vid_close: Closing video device /dev/video0
[1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
```

ccrisan, 2017-02-26:
This looks to me like a problem with the camera device. In any case, it's not motionEye-related. Make sure the V4L2 device works, go through [this](https://github.com/ccrisan/motioneye/wiki/Troubleshooting) wiki article and finally report a bug to motion.

closed 2017-02-26 by ccrisan