motioneye-project/motioneye #464: MJPEG video created on motion.. now not created

Monki77, 2017-02-28:
Yesterday i disabled "Movies" as i think it's excessive for using with the pi zero setup i have. The Zero just isn't powerful enough so i just set it to use images instead for motion capture.
I was amazed to find this morning that the camera had created a few mjpeg videos under the video section instead of the usual mp4. The is pretty good i thought... instead of creating videos it's just using the images it captured and making an mjpeg.

The only problem is after enabling video again...then disabling it, it's no longer creating the mjpegs.

I'm not sure what happened really.

Thanks

ccrisan, 2017-02-28:
I have no idea how those mjpegs were created. I don't remember adding this functionality to motionEye :)

Monki77, 2017-02-28:
Sorry I was supposed to say MPEG!
I checked the video settings and none of them give a .mpeg option. The
 closest is an mpeg4, but that still saves it as a .avi not a .mpeg, so it
can't have even been that I accidentally turned on Movie.

So this is a complete accident? Do you think it's worth me showing you the
logs from when it happened?

Thanks,
Chris

ccrisan, 2017-02-28:
Then make sure you have motion 4.0 installed and choose one of the available movie codecs.

Monki77, 2017-02-28:
I'm not sure if i was clear sorry. I'll start again.

The .mpeg video created by the motion event this morning was an actual
.mpeg file called "Front House Cam_07-34-57.mpeg". It was not any of these
file formats listed here, but even so, "Movies" was turned off so it cannot
have been anything to do with that.
[image: Inline images 1]
I am using these versions:
motionEye Version 20170212
Motion Version 4.0.1+git37b3595

I cannot even replicate the issue, but i really would like it to happen
again as the mpeg that as produced was perfect. It was as if the software
had made it's own timelapse from motion captured still images.

Is there any reason at all this would happen?

ccrisan, 2017-03-01:
It might have been created by you when you generated a timelapse from the existing pictures, using the media browser window.

Monki77, 2017-03-02:
I don't think it was, as I was asleep when the video was made. Nevermind
I'll forget it ever happened haha.

Thanks for replying.

closed 2017-08-05 by ccrisan