motioneye-project/motioneye #465: Rasbperry Pi Camera as a Network Camera in a Server

NaCMPE, 2017-02-28:
Hi, 

I was trying to achieve a network of cameras (all cameras connected to a single server), therefore, I installed MotionEyeOS on each of the raspberry pi's (where they are connected to Raspberry Pi cameras). After that, I installed MotionEye to a computer with a linux system (Ubuntu), where I can access the user interface using the localhost. However, now I am trying to connect all the Raspberry Pi devices to this server as Network Cameras but I couldn't !! 
I always get the following message over a gray screen **" Unable to open video device"**
I tried writing the URL as the streaming URL, I also tried adding the camera name at the end of the URL but nothing seems to work. 

Notes: 
- I can see the cameras by going directly from the browser to the IP of each device
- I can add these cameras as  MPJEG cameras or Remote Motioneye cameras and I can get the streaming
-  I am using only raspberry pi cameras. 

So am I missing something in the configuration ? is there a specific place that I can add the camera module to my computer (Ubuntu) so it can recognize it ? 

The help is really appreciated.

ccrisan, 2017-03-01:
The URLs to be used should be the streaming URLs. They should look something like http://192.168.1.102:8081. The rest of the setup seems fine to me.

NaCMPE, 2017-03-02:
I already used the streaming URL, but it doesn't work. Does the ssh connection between the server and the Raspberry Pi board effect this.

ccrisan, 2017-03-02:
Well does the streaming URL work when you access it directly from the browser? I mean the streaming URL (the 8081 one), not the normal motionEye URL.

hasselk, 2017-05-20:
Hi there, have the same issue. I can add motioneyeos cams as simple mjpg streaming cams without motion and stuff but when i try to add them as a network camera it doesent work. I always get " Unable to open video device" message. the motioneyeos cams run as "fast network cameras", might that be a problem?

ccrisan, 2017-05-20:
@hasselk what do you run on your "hub"? Is that another motionEyeOS?

hasselk, 2017-05-26:
@ccrisan i have an instance of motioneye running in an vm, and there i try to add my raspi cams which are running motioneyeos

ccrisan, 2017-05-27:
@hasselk Running your cameras as Fast Network Cameras shouldn't change anything - they behave as Network Cameras in the end.

I'd need to see your `motion.log` file.

JamBalaya56562, 2026-11-07:
Closing, as the `motion.log` ccrisan asked for never came, so the cause was not found. As ccrisan said above, a motionEyeOS camera is added with its streaming URL (port 8081), and that URL has to open in a browser from the server. Adding it as a simple MJPEG camera, which worked for hasselk, is another way. If a network camera still shows "Unable to open video device" with a current release, please open a new issue with `motion.log` from the server.

closed 2026-11-07 by JamBalaya56562