motioneye-project/motioneye #466: Autoreboot after Stopmotion

TheKitro, 2017-03-01:
Hello

My Motioneye Raspi reboots after some seconds (perhaps 120) after I stop the service by script.
I use the RaspiCam and LAN connection in version: 
motionEye Version 	20170212
Motion Version 	4.0.1+git37b3595

In my script I stop the service with following command:
`meyectl stopserver -b -c /etc/motioneye.conf`

I set in /data/etc/watch.conf  a line with
`meyewatch_disable="true" `
and both Network Link Watch ,	Connectivity Watch are off

my next problem is, after reboot it startet short with an detection. The motioneye seems to need 2 seconds for justage the motiondetection. my workaround is to stretch the motiondetectiontime to estimate 3 seconds.

Have someone an idea?

Thx
Cornflake

motioneye.log
```
2017-03-01 11:56:35: [motioneye]     INFO: hello! this is motionEye server 20170212
2017-03-01 11:56:37: [motioneye]     INFO: cleanup started
2017-03-01 11:56:37: [motioneye]     INFO: wsswitch started
2017-03-01 11:56:37: [motioneye]     INFO: tasks started
2017-03-01 11:56:37: [motioneye]     INFO: mjpg client garbage collector started
2017-03-01 11:56:37: [motioneye]     INFO: smb mounts started
2017-03-01 11:56:37: [motioneye]     INFO: server started
2017-03-01 12:17:21: [motioneye]     INFO: interrupt signal received, shutting down...
2017-03-01 12:17:21: [motioneye]     INFO: server stopped
2017-03-01 12:17:21: [motioneye]     INFO: tasks stopped
2017-03-01 12:17:24: [motioneye]     INFO: motion stopped
2017-03-01 12:17:24: [motioneye]     INFO: smb mounts stopped
2017-03-01 12:17:24: [motioneye]     INFO: bye!
2017-03-01 12:17:42: [motioneye]     INFO: hello! this is motionEye server 20170212
2017-03-01 12:17:45: [motioneye]     INFO: cleanup started
2017-03-01 12:17:45: [motioneye]     INFO: wsswitch started
2017-03-01 12:17:45: [motioneye]     INFO: tasks started
2017-03-01 12:17:45: [motioneye]     INFO: mjpg client garbage collector started
2017-03-01 12:17:45: [motioneye]     INFO: smb mounts started
2017-03-01 12:17:45: [motioneye]     INFO: server started
2017-03-01 12:47:08: [motioneye]     INFO: interrupt signal received, shutting down...
2017-03-01 12:47:08: [motioneye]     INFO: server stopped
2017-03-01 12:47:08: [motioneye]     INFO: tasks stopped
2017-03-01 12:47:11: [motioneye]     INFO: motion stopped
2017-03-01 12:47:11: [motioneye]     INFO: smb mounts stopped
2017-03-01 12:47:11: [motioneye]     INFO: bye!
```

message.log
```
Mar  1 11:56:35 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
Mar  1 11:56:35 meye-40320715 user.warn motion: [0:motion] [WRN] [ALL] thread config option deprecated use camera
Mar  1 11:56:35 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
Mar  1 11:56:35 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1+git37b3595 Started
Mar  1 11:56:35 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Mar  1 11:56:35 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Mar  1 11:56:35 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Mar  1 11:56:35 meye-40320715 user.warn motion: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
Mar  1 12:17:43 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
Mar  1 12:17:43 meye-40320715 user.warn motion: [0:motion] [WRN] [ALL] thread config option deprecated use camera
Mar  1 12:17:43 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
Mar  1 12:17:43 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1+git37b3595 Started
Mar  1 12:17:43 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Mar  1 12:17:43 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Mar  1 12:17:43 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Mar  1 12:17:43 meye-40320715 user.warn motion: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
```

ccrisan, 2017-03-01:
At startup, motionEye needs a few seconds indeed to communicate to motion that detection needs to be disabled. There's not much I can do about it. Sorry.

closed 2017-03-01 by ccrisan

TheKitro, 2017-03-01:
my mainproblem is the reboot. how i can stop the rebooting.

TheKitro, 2017-03-01:
The motioneye run for example 5 hours. 
Then I stop it per script. 
Then after after some time it reboots. 
Why?

TheKitro, 2017-03-01:
In the Version from okt. 2016 it worked.

ccrisan, 2017-03-01:
@TheKitro the `messages.log` file will indicate the reboot cause.

TheKitro, 2017-03-01:
Do you mean:
`Mar  1 11:56:35 meye-40320715 user.warn motion: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD`
It the same picam like the last months. I only activated the despeckle filter.

What is VIDIOC_G_STD?

ccrisan, 2017-03-01:
Nope. That excerpt from your messages log file is too small. The log line should clearly say something about *rebooting*.

TheKitro, 2017-03-02:
Hi 
Today I tested it again. after 120 seconds in stopped mode it reboots.
Have you an idea?

```
Mar  2 17:19:40 meye-40320715 auth.info sshd[3665]: Accepted password for root from 213.179.135.96 port 58678 ssh2

here I stopped the service with :
meyectl stopserver -b -c /etc/motioneye.conf

Mar  2 17:30:30 meye-40320715 user.notice motioneye: not responding for 120 seconds, rebooting
Mar  2 16:30:30 meye-40320715 daemon.info init: starting pid 4925, tty '': '/etc/init.d/rcK'
Mar  2 17:30:32 meye-40320715 auth.info sshd[745]: Received signal 15; terminating.
Mar  2 17:30:32 meye-40320715 auth.info sshd[3665]: Exiting on signal 15
Jan  1 01:00:03 meye-40320715 auth.err passwd: /etc/passwd: Read-only file system
Jan  1 01:00:03 meye-40320715 auth.err passwd: can't update password file /etc/passwd
Jan  1 01:00:03 meye-40320715 auth.err passwd: /etc/passwd: Read-only file system
Jan  1 01:00:03 meye-40320715 auth.err passwd: can't update password file /etc/passwd
Mar  2 17:30:47 meye-40320715 cron.info crond[722]: crond (busybox 1.24.2) started, log level 8
Mar  2 17:30:47 meye-40320715 cron.info crond[725]: crond (busybox 1.24.2) started, log level 8
Mar  2 17:30:48 meye-40320715 auth.info sshd[742]: Server listening on 0.0.0.0 port 22.
Mar  2 17:30:52 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
Mar  2 17:30:52 meye-40320715 user.warn motion: [0:motion] [WRN] [ALL] thread config option deprecated use camera
Mar  2 17:30:52 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
Mar  2 17:30:52 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1+git37b3595 Started
Mar  2 17:30:52 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Mar  2 17:30:52 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Mar  2 17:30:52 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Mar  2 17:30:52 meye-40320715 user.warn motion: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
Mar  2 16:30:55 meye-40320715 daemon.info init: starting pid 908, tty '/dev/ttylogin': '/sbin/getty -L ttylogin 115200 vt100'
Mar  2 17:31:10 meye-40320715 auth.info sshd[935]: WARNING: /data/etc/moduli does not exist, using fixed modulus
Mar  2 17:31:57 meye-40320715 auth.info sshd[935]: Accepted password for root from 213.179.135.96 port 58954 ssh2
Mar  2 17:35:03 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
Mar  2 17:35:03 meye-40320715 user.warn motion: [0:motion] [WRN] [ALL] thread config option deprecated use camera
Mar  2 17:35:03 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
Mar  2 17:35:03 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1+git37b3595 Started
Mar  2 17:35:03 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Mar  2 17:35:03 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Mar  2 17:35:03 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Mar  2 17:35:03 meye-40320715 user.warn motion: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
Mar  2 17:38:08 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
Mar  2 17:38:08 meye-40320715 user.warn motion: [0:motion] [WRN] [ALL] thread config option deprecated use camera
Mar  2 17:38:08 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
Mar  2 17:38:08 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1+git37b3595 Started
Mar  2 17:38:08 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Mar  2 17:38:08 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Mar  2 17:38:08 meye-40320715 user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Mar  2 17:38:08 meye-40320715 user.warn motion: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
```

ccrisan, 2017-03-02:
You need to stop it using `/etc/init.d/S85motioneye stop`. Otherwise a background checker will keep verifying for the connection and will say:
    
    motioneye: not responding for 120 seconds, rebooting

TheKitro, 2017-03-02:
Ok in place of  
meyectl stopserver -b -c /etc/motioneye.conf
I use
/etc/init.d/S85motioneye stop
and for restarting 
/etc/init.d/S85motioneye start
per script. Its ok?

What is with the line   
meyewatch_disable="true"  
in the  /data/etc/watch.conf  . 
It is obsolete?

ccrisan, 2017-03-03:
Indeed, with `meyewatch_disable="true"` in `/data/etc/watch.conf`, your system shouldn't reboot anymore. Are you sure you have rebooted after changing that setting?

TheKitro, 2017-03-03:
Last four Days I checked this Value, but today it wasn't anymore there. I retype it , saved and reboot. 
Now it works again. :)   
Have you an idea, what could be happen that this value can get lost? 
It's looks like the file was replaced by default values.
The only reason I know is after an motioneye update, but I dont updated last days.

ccrisan, 2017-03-03:
Updates won't touch your /data partition. Make sure you weren't actually playing with /etc/watch.conf instead, which *is* overwritten by updates.

TheKitro, 2017-03-06:
Thx I checked the folder. this could be my mistake.