motioneye-project/motioneye #469: motion: [1:ml1] [ERR] [ALL] init_camera_type: Unable to determine camera type (MMAL, Netcam, V4L2, BKTR)

JosephRedfern, 2017-03-02:
Hi,

I am trying to run motioneye on a Raspberry Pi (latest model) with a Microsoft Lifecam HD-3000. I have installed motion and motioneye, however, when adding the webcam through the Web UI I get an error in the web UI, and the following output to syslog:

```
Mar  2 18:46:18 raspberrypi meyectl[26050]: ERROR: motion not running, starting it
Mar  2 18:46:18 raspberrypi motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Mar  2 18:46:18 raspberrypi motion: [0:motion] [WRN] [ALL] thread config option deprecated use camera
Mar  2 18:46:18 raspberrypi motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
Mar  2 18:46:18 raspberrypi motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1+gitUNKNOWN Started
Mar  2 18:46:18 raspberrypi motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Mar  2 18:46:18 raspberrypi motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Mar  2 18:46:18 raspberrypi motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Mar  2 18:46:18 raspberrypi motion: [1:ml1] [ERR] [ALL] init_camera_type: Unable to determine camera type (MMAL, Netcam, V4L2, BKTR)
Mar  2 18:46:18 raspberrypi meyectl[26050]: ERROR: failed to start motion: motion failed to start
Mar  2 18:46:18 raspberrypi meyectl[26050]: Traceback (most recent call last):
Mar  2 18:46:18 raspberrypi meyectl[26050]: File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 323, in checker
Mar  2 18:46:18 raspberrypi meyectl[26050]: motionctl.start()
Mar  2 18:46:18 raspberrypi meyectl[26050]: File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 134, in start
Mar  2 18:46:18 raspberrypi meyectl[26050]: raise Exception('motion failed to start')
Mar  2 18:46:18 raspberrypi meyectl[26050]: Exception: motion failed to start
```

I am able to use motioneye in MJPEG mode (streaming the webcam the loopback device via vlc) but would like to be able to use motion detection etc. 

Do you have any advice? I've tried googling the issue but it appears to be unique...

Thanks!

JosephRedfern, 2017-03-02:
Hmm. I know what's causing this now -- when choosing the webcam from the UI dropdown, motioneye used `/dev/v4l/by-id/usb-Microsoft_Microsoft®_LifeCam_HD-3000-video-index0`. However, motion expects v4l devices to be specified in the form`/dev/videoN` (https://github.com/Motion-Project/motion/blob/bde92a41863615146f9d0469d5b4db849962b935/motion.c#L724).

Manually editing thread-1.conf to specify `/dev/video0` resolved this problem, but perhaps it's something that needs looking at?

ccrisan, 2017-03-03:
This is odd. I've been using "persistent" camera device paths for a long time without any issues. @Mr-Dave, is this camera device filename matching something you guys have introduced recently to motion?

Mr-Dave, 2017-03-03:
This issue has been previously logged and the patch has been built and will be pulled in shortly to the Motion code.

JosephRedfern, 2017-03-03:
@Mr-Dave, @ccrisan Great, thanks for getting back to me. Sorry for the duplicate issue (I wasn't actually sure if this was a Motion or MotionEye problem) -- like I said, I had tried to find existing reports but hadn't found anything. I guess this issue can be closed?

ccrisan, 2017-03-03:
@JosephRedfern I haven't been aware of this either until these days. Anyways, @Mr-Dave thanks for getting back to us with an answer.

closed 2017-03-03 by ccrisan