motioneye-project/motioneye #471: arch linux antergos installed ok but An error occurred. Refreshing is recommended

francisuk1989, 2017-03-06:
I have installed antergos based on arch linux and installed motioneye fine from the Wiki. However i am setting up an h264 IP Camera via rtsp://192.168.254.19/12 but it wont let me view the IP Camera and comes up as an error,

An error occurred. Refreshing is recommended 

motion.log
[11535680:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[11535680:motion] [WRN] [ALL] thread config option deprecated use camera
[11535680:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[11535680:motion] [NTC] [ALL] motion_startup: Motion 4.0.1 Started
[11535680:motion] [NTC] [ALL] motion_startup: Logging to syslog
[11535680:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[11535680:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.254.19/12)

ccrisan, 2017-03-06:
`Exception: motion failed to start` - it would be useful to see what motion has to say.

francisuk1989, 2017-03-07:
[francis@archysrv ~]$ nano /var/log/motion.log
  GNU nano 2.7.5                                          File: /var/log/motion.log
[22025536:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[22025536:motion] [WRN] [ALL] thread config option deprecated use camera
[22025536:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[22025536:motion] [NTC] [ALL] motion_startup: Motion 4.0.1 Started
[22025536:motion] [NTC] [ALL] motion_startup: Logging to syslog
[22025536:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[22025536:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://176.130.11.204/12)

ccrisan, 2017-03-07:
The log doesn't show too much information. Is this the full log?

francisuk1989, 2017-03-07:
/var/log/motion.log ??
Will set to Full log.

francisuk1989, 2017-03-07:
Yes, nano /var/log/motion.log only shows the above infomation, I cant see no logs from motioneye iself

francisuk1989, 2017-03-09:
I have had a play with Diffrents Distros based upon Arch Linyx but nothing seems to work.

Maybe an Motion issue

Hope someone can shine a light on a fix
https://aur.archlinux.org/packages/motioneye/?comments=all
https://forum.manjaro.org/t/motioneye-motion-failed-to-start/19114

ccrisan, 2017-03-09:
@francisuk1989 try isolating the problem by running motion manually with the configuration files. If your config files are in `/etc`, as indicated by the wiki instructions, then you should do something like:

    cd /etc
    motion -n -d 9 -c motion.conf

See how this works out. If it fails, open an issue on the motion project by attaching your motion config files (`motion.conf`, `thread-1.conf`) and log file, *after* removing any sensitive data.

francisuk1989, 2017-03-10:
@ccrisan will give it ago, I have tryed Ubuntu and everything seems to work fine.

schmeckles22, 2017-05-23:
I can confirm I also have this issue on Antergos

francisuk1989, 2017-06-16:
ShatterGlass commented on 2017-06-06 21:47
francisuk1989 I had the same problem with motion and my CPU. **Installing motion-git from AUR fix the problem for me.**

if this any help??

ioquatix, 2017-09-10:
I'm not sure if this is the same problem but I'm getting

```
hana.oriontransfer.co.nz meyectl[2836]:    ERROR: motion not running, starting it
Sep 11 03:06:46 hana.oriontransfer.co.nz motion[3985]: [23663936:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Sep 11 03:06:46 hana.oriontransfer.co.nz motion[3985]: [23663936:motion] [WRN] [ALL] thread config option deprecated use camera
Sep 11 03:06:46 hana.oriontransfer.co.nz motion[3985]: [23663936:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
Sep 11 03:06:46 hana.oriontransfer.co.nz motion[3985]: [23663936:motion] [NTC] [ALL] motion_startup: Motion 4.0.1 Started
Sep 11 03:06:46 hana.oriontransfer.co.nz motion[3985]: [23663936:motion] [NTC] [ALL] motion_startup: Logging to syslog
Sep 11 03:06:46 hana.oriontransfer.co.nz motion[3985]: [23663936:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Sep 11 03:06:46 hana.oriontransfer.co.nz motion[3985]: [23663936:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Sep 11 03:06:46 hana.oriontransfer.co.nz motion[3985]: [1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.20:554/live0.264)
Sep 11 03:06:46 hana.oriontransfer.co.nz kernel: traps: ml1[3986] trap invalid opcode ip:414fce sp:7fb496375b10 error:0 in motion[400000+40000]
Sep 11 03:06:46 hana.oriontransfer.co.nz systemd[1]: Started Process Core Dump (PID 3988/UID 0).
Sep 11 03:06:47 hana.oriontransfer.co.nz meyectl[2836]:    ERROR: failed to start motion: motion failed to start
Sep 11 03:06:47 hana.oriontransfer.co.nz meyectl[2836]: Traceback (most recent call last):
Sep 11 03:06:47 hana.oriontransfer.co.nz meyectl[2836]:   File "/usr/lib/python2.7/site-packages/motioneye/server.py", line 323, in checker
Sep 11 03:06:47 hana.oriontransfer.co.nz meyectl[2836]:     motionctl.start()
Sep 11 03:06:47 hana.oriontransfer.co.nz meyectl[2836]:   File "/usr/lib/python2.7/site-packages/motioneye/motionctl.py", line 131, in start
Sep 11 03:06:47 hana.oriontransfer.co.nz meyectl[2836]:     raise Exception('motion failed to start')
Sep 11 03:06:47 hana.oriontransfer.co.nz meyectl[2836]: Exception: motion failed to start
Sep 11 03:06:47 hana.oriontransfer.co.nz systemd-coredump[3989]: Process 3985 (motion) of user 0 dumped core.
                                                                 
                                                                 Stack trace of thread 3986:
                                                                 #0  0x0000000000414fce n/a (motion)
                                                                 #1  0x0000000000411a00 n/a (motion)
                                                                 #2  0x000000000040814d n/a (motion)
                                                                 #3  0x00007fb4a5819049 start_thread (libpthread.so.0)
                                                                 #4  0x00007fb4a2fbef0f __clone (libc.so.6)
                                                                 
                                                                 Stack trace of thread 3985:
                                                                 #0  0x00007fb4a5822d8d __nanosleep (libpthread.so.0)
                                                                 #1  0x0000000000405823 n/a (motion)
                                                                 #2  0x00007fb4a2ef14ca __libc_start_main (libc.so.6)
                                                                 #3  0x0000000000405eba n/a (motion)
                                                                 
                                                                 Stack trace of thread 3987:
                                                                 #0  0x00007fb4a2fb6fd3 __select (libc.so.6)
                                                                 #1  0x0000000000423bea n/a (motion)
                                                                 #2  0x0000000000423d2e n/a (motion)
                                                                 #3  0x00007fb4a5819049 start_thread (libpthread.so.0)
                                                                 #4  0x00007fb4a2fbef0f __clone (libc.so.6)
```

I will try motion-git.

ioquatix, 2017-09-10:
I installed `motion-git` from AUR and got the following:

```
Sep 11 03:14:40 hana.oriontransfer.co.nz motion[26299]: [1:ml1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Sep 11 03:14:40 hana.oriontransfer.co.nz motion[26299]: [1:ml1] [NTC] [VID] vid_start: Opening V4L2 device
Sep 11 03:14:40 hana.oriontransfer.co.nz motion[26299]: [1:ml1] [NTC] [VID] v4l2_start: Using videodevice /dev/video0 and input -1
Sep 11 03:14:40 hana.oriontransfer.co.nz motion[26299]: [1:ml1] [ALR] [VID] v4l2_start: Failed to open video device /dev/video0: No such file or directory
Sep 11 03:14:40 hana.oriontransfer.co.nz motion[26299]: [1:ml1] [ERR] [VID] vid_start: V4L2 device failed to open
Sep 11 03:14:45 hana.oriontransfer.co.nz meyectl[2796]:    ERROR: motion not running, starting it
Sep 11 03:14:45 hana.oriontransfer.co.nz motion[26787]: [532912064:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Sep 11 03:14:45 hana.oriontransfer.co.nz motion[26787]: [532912064:motion] [ALR] [ALL] conf_cmdparse: Deprecated config option "thread" since after version 3.4.1:
Sep 11 03:14:45 hana.oriontransfer.co.nz motion[26787]: [532912064:motion] [ALR] [ALL] conf_cmdparse: The "thread" option has been replaced by the "camera" option.
Sep 11 03:14:45 hana.oriontransfer.co.nz motion[26787]: [532912064:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
Sep 11 03:14:45 hana.oriontransfer.co.nz motion[26787]: [532912064:motion] [NTC] [ALL] motion_startup: Motion 4.0.1+giteeecf61 Started
Sep 11 03:14:45 hana.oriontransfer.co.nz motion[26787]: [532912064:motion] [NTC] [ALL] motion_startup: Logging to syslog
Sep 11 03:14:45 hana.oriontransfer.co.nz motion[26787]: [532912064:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Sep 11 03:14:45 hana.oriontransfer.co.nz motion[26787]: [532912064:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Sep 11 03:14:47 hana.oriontransfer.co.nz motion[26787]: [1:ml1] [CRT] [STR] http_bindsock: error binding on 127.0.0.1 port 8081: Address already in use
Sep 11 03:14:47 hana.oriontransfer.co.nz kernel: ml1[26788]: segfault at 10 ip 00007f819d7980e3 sp 00007f818c670af0 error 4 in libc-2.26.so[7f819d6e3000+1ae000]
Sep 11 03:14:47 hana.oriontransfer.co.nz motion[26787]: [1:ml1] [ERR] [ALL] motion_init: Problem enabling motion-stream server in port 8081: Address already in use
Sep 11 03:14:47 hana.oriontransfer.co.nz systemd[1]: Started Process Core Dump (PID 26872/UID 0).
Sep 11 03:14:48 hana.oriontransfer.co.nz systemd-coredump[26873]: Process 26787 (motion) of user 0 dumped core.
                                                                  
                                                                  Stack trace of thread 26788:
                                                                  #0  0x00007f819d7980e3 __tz_convert (libc.so.6)
                                                                  #1  0x0000564c1e6e4cc0 n/a (motion)
                                                                  #2  0x0000564c1e6fa1b3 n/a (motion)
                                                                  #3  0x0000564c1e6fbc79 n/a (motion)
                                                                  #4  0x0000564c1e6e3a7c n/a (motion)
                                                                  #5  0x0000564c1e6e80b9 n/a (motion)
                                                                  #6  0x00007f819daa108a start_thread (libpthread.so.0)
                                                                  #7  0x00007f819d7d91bf __clone (libc.so.6)
                                                                  
                                                                  Stack trace of thread 26787:
                                                                  #0  0x00007f819daab230 __nanosleep (libpthread.so.0)
                                                                  #1  0x0000564c1e6e2c54 main (motion)
                                                                  #2  0x00007f819d703f6a __libc_start_main (libc.so.6)
                                                                  #3  0x0000564c1e6e337a n/a (motion)
                                                                  
                                                                  Stack trace of thread 26871:
                                                                  #0  0x00007f819e96e06f n/a (libavcodec.so.57)
                                                                  #1  0x6f71737577757676 n/a (n/a)
```

ioquatix, 2017-09-10:
Okay it was trying to run both the old and new system at the same time in the last update. I've fixed it and it's working fine now, using `motion-git`.

closed 2017-09-11 by ccrisan

francisuk1989, 2017-10-04:
When you install the 

> motion-git

 it tells you to remove motion

> :: motion-git and motion are in conflict. Remove motion? [y/N] y
> 
> Packages (2) motion-4.0.1-2 [removal]  motion-git-4.0.r273.g2eedfdc-1

@ccrisan can you update the Wiki for installion? or link it back to here? may help someone out there.

ccrisan, 2017-10-05:
@francisuk1989 what do you want me to update, more precisely? Of course motion-git and motion are in conflict.

francisuk1989, 2017-10-05:
@ccrisan  In the Arch installation https://github.com/ccrisan/motioneye/wiki/Install-On-Arch

1)
pacman -S ffmpeg v4l-utils **pacaur**
(This will pull ffmpeg, v4l-utils  as nornmal but includes pacaur for AUR)

pacaur -a motion-git
(This will pull motion-git from AUR https://aur.archlinux.org/packages/motion-git without this motioneye will give an error)

the reset of the steps are working fine.