motioneye-project/motioneye #475: motionEye without network connection

Henrietta1989, 2017-03-07:
Is there any way to use motionEye in Raspberry Pi 3 which has only two USB cams and maybe one ethernet, cam without keeping internet connection on?

This type of setups are used a lot in places without possibility to have network connection.

Now if i close internet connection and open browser, it dont find motionEye webpage at all.

Thanks

ccrisan, 2017-03-07:
I believe you're referring to motionEyeOS. If that is the case, see [this](https://github.com/ccrisan/motioneyeos/wiki/FAQ#how-can-i-run-motioneyeos-without-a-network-connection).

closed 2017-03-07 by ccrisan

Henrietta1989, 2017-03-07:
Actually it is motionEyes not motionEyesOS. Thanks

ccrisan, 2017-03-07:
If it's motionEye, then it has nothing to do with the network connection.