motioneye-project/motioneye #479: Failed to connect wifi cam RaspberryPi Zero W

HJSK, 2017-03-09:
I'm newbee to motioneye. 
I have motionEye (2017 feb) installed on raspberryPi 3b
Connected csi CAM on this Pi
Connected USB cam also on this pi
Now I configured new Pi zero W als with CSI cam
By trying to connect this pi zero W board to motionEye I got the message 
Failed to connect wifi cam RaspberryPi Zero W. 
With ssh I can connect to pi zero W without problem. 

What can possible wrong ? 

Thanks in advantage for support!

Harry Jeurink
Slovakia

[image]
[image]

ccrisan, 2017-03-09:
Assuming that 192.168.1.17 is the IP address of your Pi Zero W, make sure it runs motionEye on port 8765 (it doesn't appear to do so now). Also I doubt the user `pi` is indeed the motionEye user configured on your Zero (it should be `admin` if you haven't deeply tweaked stuff there).

HJSK, 2017-03-10:
Thx for your quick reply! 
I did not understand that using a raspberry with wifi and CAM should also run motioneye. 
My idea that only the server should run motionEye. 
I will install MotionEye also on the pi zero W and come back with result.

HJSK, 2017-03-12:
I tried to install 20170212 version for zero (zero W is different, because of WiFi and bluetooth on the board).
This is not succesfull because the image on the SDHC card tries to install  wired network which is not on it. 
I also tried (for testing) to ihnstall 20170212 Pi2 and Pi 3 on the Pi Zero W.
Also they cannot install MotionEye from scratch.
What else can I do?  

Hope somebody got an idea. 

Thanks in advantage!

JamBalaya56562, 2026-11-09:
As ccrisan explained, the other Pi has to run motionEye too, to be added as a remote motionEye camera. The image problem on the Zero W was one of motionEyeOS images from 2017, which have long been replaced. Closing.

closed 2026-11-09 by JamBalaya56562