motioneye-project/motioneye #481: Web interface missing settings?

MrGlasspoole, 2017-03-11:
I did install motionEye in Debian on my Hyper-V server by this instruction:
https://github.com/ccrisan/motioneyeos/issues/333

On a Pi3 i have set motionEyeOS to Fast Network Camera.
I added the Pi as Network Camera with http://10.1.255.16:8081/

But i don't have all the settings i have on the Pi if it is not set to Fast Network Camera.
I can only turn on/off Video Streaming (does not expand).
No "Text Overlay", no options like "Interval Snapshots" or options in Movies.
Only the option for Preserve time in both.
I don't see something to disable Motion Detection (something i don't need)?

Also i wonder on a full Linux install i can't do picture capturing or video recording if i add a camera as Simple MJPEG Camera?

Here
https://www.bountysource.com/issues/32149973-rpi-3-motioneye-slow-preview-recording-fps
i did read that 9 cams work with a Odroid XU4 but i have now just one cam on a i3 and have just 4 fps?

ccrisan, 2017-03-11:
That's because you added your camera as a Simple MJPEG Camera. Add it as a normal Network Camera and you'll get your options back.

closed 2017-03-11 by ccrisan

MrGlasspoole, 2017-03-11:
> I added the Pi as Network Camera with http://10.1.255.16:8081/

ccrisan, 2017-03-11:
So what is the option that you're missing now?

MrGlasspoole, 2017-03-11:
As i wrote:
No "Text Overlay", no options like "Interval Snapshots" or options in Movies.
I can't expand Video Streaming.
This is all i have:
[image]

ccrisan, 2017-03-11:
Hmm you did not enable Advanced Settings, did you?

MrGlasspoole, 2017-03-11:
I can't believe i did miss this setting :-(
Thanks you!

This brings up some questions now.
Is it better to directly use the Pi's streaming URL (if i don't care for overlay's)
or the one from the server - from a performance point of view?

Then i have now 3 places i can set the frame rate.
I guess the frame rate on the Pi must be equal or higher then the rates on the server?

ccrisan, 2017-03-11:
You have too many questions here to be answered in a single issue (which wasn't opened for any of them in the first place). Anyways, I'll give it a go:

The FNC mode makes your PI behave just like any other regular network camera - nothing motionEye-specific. Use the provided UI on your PI to tweak the frame rate, resolution and settings so that you're happy with your netcam as it is. You can use your browser to directly test the mjpeg stream.

Now, when you add it to your motionEye server, you'll have to tweak framerates again. Depending on how you want to use it, you may want a lower framerate on the server side, or you may go just as high as you set your FNC. Keep in mind though that the server does some processing and that you may not reach the target framerate.

MrGlasspoole, 2017-03-12:
Sorry, i don't have much experience writing on GitHub :-(
What do you mean by "which wasn't opened for any of them in the first place"?

imjoeco, 2018-06-21:
@ccrisan I had set up motion eye and everything was working fine. After resetting my network and rebooting the rPi, everything's still functioning as before but the camera feeds no longer have a link to their settings, and the main settings menu is missing pretty much everything. Couldn't find an open issue for this but this one seemed similar enough so I figured I'd post it here as an FYI.

[image]
[image]

ccrisan, 2018-06-21:
Log out and log in as admin

RunTime22, 2019-08-14:
> As i wrote:
> No "Text Overlay", no options like "Interval Snapshots" or options in Movies.
> I can't expand Video Streaming.
> This is all i have:
> [image]
Hello,
I have the same problem, did you fix it?

I have a spy-camera v.1.0 and a raspberry pi zero connected with LAN cable.

Thanks
RT