motioneye-project/motioneye #482: Maximum number of USB webcams ?

MacAre, 2017-03-11:
Hello,
I have a raspberry PI3 with motioneye and RPI-Monitor, on a 32GB card
How many webcam can connect?
With one, I have a good throughput (10 fps without problems).
With two, I have to drop to 3 fps.
With a third, one of the three jumps ...
The three webcam are in 640x480

I bought 3 identical webcam because the first works very well alone.
http://www.electrodepot.fr/webcam-micro-apm-s-059.html

What is the recommended driver ?
uvcvideo (YUYV) or gspca_zc3xx (JPEG)
APM S059 is uvcvideo

```
uname -a
Linux PI3_Nico 4.9.13-v7+ #974 SMP Wed Mar 1 20:09:48 GMT 2017 armv7l GNU/Linux
```
motionEye Version 	0.35.2
Motion Version 	4.0.1

Syslog with 3 connected webcam:
```
Mar 11 23:21:30 PI3_Nico motion: [3:ml3] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
Mar 11 23:21:40 PI3_Nico motion: [3:ml3] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Mar 11 23:21:40 PI3_Nico motion: [3:ml3] [ERR] [VID] v4l2_select_input: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
Mar 11 23:21:40 PI3_Nico motion: [3:ml3] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
Mar 11 23:21:50 PI3_Nico motion: [3:ml3] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Mar 11 23:21:50 PI3_Nico motion: [3:ml3] [ERR] [VID] v4l2_select_input: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
Mar 11 23:21:50 PI3_Nico motion: [3:ml3] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
Mar 11 23:22:00 PI3_Nico motion: [3:ml3] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Mar 11 23:22:00 PI3_Nico motion: [3:ml3] [ERR] [VID] v4l2_select_input: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
Mar 11 23:22:00 PI3_Nico motion: [3:ml3] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
Mar 11 23:22:10 PI3_Nico motion: [3:ml3] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Mar 11 23:22:10 PI3_Nico rsyslogd-2007: action 'action 17' suspended, next retry is Sat Mar 11 23:23:40 2017 [try http://www.rsyslog.com/e/2007 ]
Mar 11 23:22:10 PI3_Nico motion: [3:ml3] [ERR] [VID] v4l2_select_input: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
Mar 11 23:22:10 PI3_Nico motion: [3:ml3] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
Mar 11 23:22:20 PI3_Nico motion: [3:ml3] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Mar 11 23:22:20 PI3_Nico motion: [3:ml3] [ERR] [VID] v4l2_select_input: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
Mar 11 23:22:20 PI3_Nico motion: [3:ml3] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
```

```
lsusb
Bus 001 Device 016: ID 1908:2310 GEMBIRD
Bus 001 Device 014: ID 1908:2310 GEMBIRD
Bus 001 Device 017: ID 1908:2310 GEMBIRD
Bus 001 Device 013: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 012: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
v4l2-ctl --list-devices
USB2.0 PC CAMERA (usb-3f980000.usb-1.2):
        /dev/video1

USB2.0 PC CAMERA (usb-3f980000.usb-1.3):
        /dev/video0

USB2.0 PC CAMERA (usb-3f980000.usb-1.5):
        /dev/video2
```
```
v4l-info /dev/video0

### v4l2 device info [/dev/video0] ###
general info
    VIDIOC_QUERYCAP
        driver                  : "uvcvideo"
        card                    : "USB2.0 PC CAMERA"
        bus_info                : "usb-3f980000.usb-1.3"
        version                 : 4.9.13
        capabilities            : 0x84200001 [VIDEO_CAPTURE,?,STREAMING,(null)]

standards

inputs
    VIDIOC_ENUMINPUT(0)
        index                   : 0
        name                    : "Camera 1"
        type                    : CAMERA
        audioset                : 0
        tuner                   : 0
        std                     : 0x0 []
        status                  : 0x0 []

video capture
    VIDIOC_ENUM_FMT(0,VIDEO_CAPTURE)
        index                   : 0
        type                    : VIDEO_CAPTURE
        flags                   : 0
        description             : "YUYV 4:2:2"
        pixelformat             : 0x56595559 [YUYV]
    VIDIOC_G_FMT(VIDEO_CAPTURE)
        type                    : VIDEO_CAPTURE
        fmt.pix.width           : 640
        fmt.pix.height          : 480
        fmt.pix.pixelformat     : 0x56595559 [YUYV]
        fmt.pix.field           : NONE
        fmt.pix.bytesperline    : 1280
        fmt.pix.sizeimage       : 614400
        fmt.pix.colorspace      : unknown
        fmt.pix.priv            : 4276996862

controls
    VIDIOC_QUERYCTRL(BASE+0)
        id                      : 9963776
        type                    : INTEGER
        name                    : "Brightness"
        minimum                 : 0
        maximum                 : 255
        step                    : 1
        default_value           : -8193
        flags                   : 0
    VIDIOC_QUERYCTRL(BASE+1)
        id                      : 9963777
        type                    : INTEGER
        name                    : "Contrast"
        minimum                 : 0
        maximum                 : 255
        step                    : 1
        default_value           : 57343
        flags                   : 0
    VIDIOC_QUERYCTRL(BASE+2)
        id                      : 9963778
        type                    : INTEGER
        name                    : "Saturation"
        minimum                 : 0
        maximum                 : 255
        step                    : 1
        default_value           : 57343
        flags                   : 0
    VIDIOC_QUERYCTRL(BASE+3)
        id                      : 9963779
        type                    : INTEGER
        name                    : "Hue"
        minimum                 : -127
        maximum                 : 127
        step                    : 1
        default_value           : -8193
        flags                   : 0
    VIDIOC_QUERYCTRL(BASE+16)
        id                      : 9963792
        type                    : INTEGER
        name                    : "Gamma"
        minimum                 : 1
        maximum                 : 8
        step                    : 1
        default_value           : 57343
        flags                   : 0
    VIDIOC_QUERYCTRL(BASE+24)
        id                      : 9963800
        type                    : MENU
        name                    : "Power Line Frequency"
        minimum                 : 0
        maximum                 : 2
        step                    : 1
        default_value           : 1
        flags                   : 0
    VIDIOC_QUERYCTRL(BASE+27)
        id                      : 9963803
        type                    : INTEGER
        name                    : "Sharpness"
        minimum                 : 0
        maximum                 : 15
        step                    : 1
        default_value           : 57343
        flags                   : 0
    VIDIOC_QUERYCTRL(BASE+28)
        id                      : 9963804
        type                    : INTEGER
        name                    : "Backlight Compensation"
        minimum                 : 1
        maximum                 : 5
        step                    : 1
        default_value           : 57343
        flags                   : 0
```
Note: MotionEye is a great job. Thank you ;-)

MacAre, 2017-03-11:
It works better when I set up a webcam in 640x480 and the other two sen 320x240

MacAre, 2017-04-05:
Hello ;-)

I increased the graphics memory to 256MB
I disable the launch of the graphical interface at launch, and I passed the resolution of the Raspberry at 640x480.
The 3 webcams work a little better, but MotionEyes loses them regularly.

Complement:
```
Apr  5 01:47:06 PI3_1_Nico kernel: [ 7249.740956] smsc95xx 1-1.1:1.0 eth0: Failed to read reg index 0x00000114: -110
Apr  5 01:47:06 PI3_1_Nico kernel: [ 7249.740971] smsc95xx 1-1.1:1.0 eth0: Error reading MII_ACCESS
Apr  5 01:47:06 PI3_1_Nico kernel: [ 7249.740980] smsc95xx 1-1.1:1.0 eth0: MII is busy in smsc95xx_mdio_read
Apr  5 01:47:06 PI3_1_Nico kernel: [ 7249.740988] smsc95xx 1-1.1:1.0 eth0: Failed to read MII_BMSR
Apr  5 01:47:06 PI3_1_Nico rsyslogd-2007: action 'action 17' suspended, next retry is Wed Apr  5 01:48:36 2017 [try http://www.rsyslog.com/e/2007 ]
Apr  5 01:47:11 PI3_1_Nico meyectl[464]: ERROR: mjpg client timed out receiving data for camera 1 on port 8081
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: ERROR: Exception in callback <functools.partial object at 0x75161de0>
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: Traceback (most recent call last):
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: File "/usr/local/lib/python2.7/dist-packages/tornado/ioloop.py", line 604, in _run_callback
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: ret = callback()
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: File "/usr/local/lib/python2.7/dist-packages/tornado/stack_context.py", line 275, in null_wrapper
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: return fn(*args, **kwargs)
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: File "/usr/local/lib/python2.7/dist-packages/motioneye/mjpgclient.py", line 306, in _garbage_collector
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: motionctl.stop(invalidate=True) # this will close all the mjpg clients
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 190, in stop
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: raise Exception('could not terminate the motion process')
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: Exception: could not terminate the motion process
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: WARNING: Connect error on fd 16: ECONNREFUSED
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: WARNING: Connect error on fd 17: ECONNREFUSED
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: WARNING: Connect error on fd 15: ECONNREFUSED
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: WARNING: skipping motion detection status update for camera with id 4: Failed to connect to 127.0.0.1 port 7999: Connection refused
Apr  5 01:47:18 PI3_1_Nico meyectl[464]: WARNING: Connect error on fd 16: ECONNREFUSED
Apr  5 01:47:19 PI3_1_Nico meyectl[464]: WARNING: Connect error on fd 14: ECONNREFUSED
Apr  5 01:47:19 PI3_1_Nico meyectl[464]: WARNING: Connect error on fd 16: ECONNREFUSED
Apr  5 01:47:19 PI3_1_Nico meyectl[464]: WARNING: Connect error on fd 15: ECONNREFUSED
```

ccrisan, 2017-04-05:
That sounds like a hardware issue (such as USB power-related problems). It's not something to fix about motionEye.

closed 2017-04-05 by ccrisan

MacAre, 2017-04-06:
I managed to correct the problems.
The 5v power supply was a bit long (6m), I put a 470μF capacitor near the Raspberry. (I will put a 2200μF).
The CPU was also very hot, I removed the converter (there is already a radiator on the CPU).
MotionEye has been running for more than 24 hours without a problem.

Thank you for responding, it helped me to search.