motioneye-project/motioneye #484: Can't connect to rtsp feed

chumy76, 2017-03-12:
I have installed motioeyeos on a rpi3, this works fine.  I have done a manual install following instructions on a rpi3, this also worked fine.

I have followed instructions to install on debian and ubuntu (both latest distro) and cannot get the video feed to work on any of them.  I know the connection info is correct as still running on the rpi3.

Logs show:

code/ [2:ml2] [DBG] [ALL] motion_loop: vid_return_code 1
[1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.2:554/VideoMain)
[1:ml1] [DBG] [NET] netcam_url_parse: Entry netcam_url_parse data rtsp://192.168.1.2:554/VideoMain
[1:ml1] [DBG] [NET] netcam_url_parse: Parse case 0 data rtsp://192.168.1.2:554/VideoMain
[1:ml1] [DBG] [NET] netcam_url_parse: Parse case 1 data rtsp
[1:ml1] [DBG] [NET] netcam_url_parse: Parse case 6 data 192.168.1.2
[1:ml1] [DBG] [NET] netcam_url_parse: Parse case 7 data :554
[1:ml1] [DBG] [NET] netcam_url_parse: Parse case 8 data 554
[1:ml1] [DBG] [NET] netcam_url_parse: Parse case 9 data /VideoMain
[1:ml1] [INF] [NET] netcam_start: Netcam_http parameter 'off' converts to flags: HTTP/1.0: 1 HTTP/1.1: 0 Keep-Alive O
[1:ml1] [INF] [NET] netcam_start: now calling netcam_setup_rtsp()
[1:ml1] [NTC] [NET] netcam_rtsp_open_context: Using tcp transport
[2:ml2] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[2:ml2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.2:554/VideoMain)
[2:ml2] [DBG] [NET] netcam_url_parse: Entry netcam_url_parse data rtsp://192.168.1.2:554/VideoMain
[2:ml2] [DBG] [NET] netcam_url_parse: Parse case 0 data rtsp://192.168.1.2:554/VideoMain
[2:ml2] [DBG] [NET] netcam_url_parse: Parse case 1 data rtsp
[2:ml2] [DBG] [NET] netcam_url_parse: Parse case 6 data 192.168.1.2
[2:ml2] [DBG] [NET] netcam_url_parse: Parse case 7 data :554
[2:ml2] [DBG] [NET] netcam_url_parse: Parse case 8 data 554
[2:ml2] [DBG] [NET] netcam_url_parse: Parse case 9 data /VideoMain
[2:ml2] [INF] [NET] netcam_start: Netcam_http parameter 'off' converts to flags: HTTP/1.0: 1 HTTP/1.1: 0 Keep-Alive O
[2:ml2] [INF] [NET] netcam_start: now calling netcam_setup_rtsp()
[2:ml2] [NTC] [NET] netcam_rtsp_open_context: Using tcp transport
[1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 500 Internal Server Error
[1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://usr:pwd@192.168.1.2:554/VideoMainver returned 5XX Server Error reply

I've removed user a pwd from log above.  

As I said I've tried on debian and ubuntu on xenserver.  I get the UI ok but cannot get a stream to work.  

When I enter cam url it detects it as a valid cam.  

I have also tried to update motion to the latest but still no joy.  
Cheers for any advice :-)

ccrisan, 2017-03-12:
Make sure you're running motion 4.x. The line that looks suspicious to me is:

    [1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 500 Internal Server Error

Anyways, please report this to [motion](https://github.com/Motion-Project/motion) as it has nothing to do with motionEye.

closed 2017-03-12 by ccrisan

chumy76, 2017-03-12:
OK, thanks.  BTW using the latest motion.