motioneye-project/motioneye #486: Request: FTP upload scheduled and motion images

m4rky-m4rk, 2017-03-12:
Hi,

I would like to FTP upload images to my website . Both on a schedule at regular intervals and also extra uploads should any motion occur. I already do this with my other IP cameras and would like MotionEye to do this too. 

This very convenient. I just check my web page and see all images from all cameras. Usually its  just scheduled half hourly shots which means all is running OK.

If FTP upload is not possible could MotionEye save images to a drive on a schedule AND also on motion. It seems that I cannot do both.

Thanks, Mark

maxholgasson, 2017-03-12:
Oh that would be great!!!
I spent now the entire evening trying to upload images taken by interval to a ftp server without success..  would be great

m4rky-m4rk, 2017-03-13:
I should have posted this on MotioneyeOS !!

maxholgasson, 2017-03-13:
I would appreciate that for the regular motioneye not for the OS one

maxholgasson, 2017-04-05:
Are there are any plans to implement that in the nearer future into the regular motioneye?
Would be great :)

ccrisan, 2017-06-11:
https://github.com/ccrisan/motioneye/commit/296c29714ce2ab8deb0c9461b074ef39493b770a adds support for uploading to FTP servers.

closed 2017-06-11 by ccrisan

sv-kov, 2017-06-14:
The test connection to the FTP server passes. Photo is done once a minute. And on the FTP server, nothing is sent. What did I not include or configured?
Thank you in advance.

sv-kov, 2017-06-15:
I can not send photos to FTP. Can I give a brief instruction on how to do this?

ccrisan, 2017-06-15:
Does your motioneye log show any errors?