motioneye-project/motioneye #487: RPI 2 Camera quality

jtx99, 2017-03-12:
When I installed the pre-built version of motioneye and the linux OS, the camera video feed was clear as a bell.

Now that I did it manually, the camera video is washed out and over-exposed. I've tried the slide bars, but its terrible.

I am using the sudo modprobe bcm2835-v4l2 command to get video from this PI camera.

Is the pre-built installation using a different video mode or driver?

If so, how do I change it.  It's terrible as it is now.

kurazli, 2017-05-08:
what resolution are using right now?
i have the same issue with high resolutions, the picture will become to bright.
with smaller resolutions it is working just fine

jtx99, 2017-05-08:
I ended up going back to the pre-built image that Calin Crisan offers as I couldnt fix the problem with video on my own.

I tried all the resolutions listed and none of them fixed the bad quality.

I tend to think its the video USB driver but could never figure out what one was being used in the pre made install so I just went back to the motioneye system and it works.

I wanted the FTP option, but so far its not happening.

JamBalaya56562, 2026-11-07:
Closing, as you went back to the prebuilt image, which worked, and the washed-out picture with the manual install, then through the bcm2835-v4l2 driver, was never tracked down. On current Raspberry Pi OS the camera module goes through libcamera, see [Troubleshooting](https://github.com/motioneye-project/motioneye/wiki/Troubleshooting#libcamera-camera-stack). Please open a new issue if the picture is still overexposed with a current release.

closed 2026-11-07 by JamBalaya56562