motioneye-project/motioneye #490: get_oformat: ffmpeg_video_codec option value mkv (and mp4) is not supported errors

entmike, 2017-03-12:
Hey guys,

Installed motioneye on Ubuntu server.  Everything works great except I cannot save video in MP4 or MKV formats.  I keep seeing this entry over and over in the logs.

I've tried to Google for it, it just keeps bringing me results that don't seem to be my issue.  I followed the Ubuntu installation guide to the letter as far as I can tell, though.

Any thoughts on what I should try?  I tried installing VLC packages and  ubuntu-restricted-extras as a try but no difference...

```
mike@ubuntu:~$ journalctl -f | grep ffmpeg
Mar 12 13:05:48 ubuntu motion[95524]: [1] [ERR] [ENC] get_oformat: ffmpeg_video_codec option value mkv is not supported
Mar 12 13:05:48 ubuntu motion[95524]: [1] [ERR] [EVT] event_ffmpeg_newfile: ffopen_open error creating (new) file [/var/lib/motioneye/Camera2/2017-03-12/13-05-48]: Operation now in progress
Mar 12 13:05:51 ubuntu motion[95524]: [1] [ERR] [ENC] get_oformat: ffmpeg_video_codec option value mkv is not supported
Mar 12 13:05:51 ubuntu motion[95524]: [1] [ERR] [EVT] event_ffmpeg_newfile: ffopen_open error creating (new) file [/var/lib/motioneye/Camera2/2017-03-12/13-05-51]: Operation now in progress
Mar 12 13:05:54 ubuntu motion[95524]: [1] [ERR] [ENC] get_oformat: ffmpeg_video_codec option value mkv is not supported
Mar 12 13:05:54 ubuntu motion[95524]: [1] [ERR] [EVT] event_ffmpeg_newfile: ffopen_open error creating (new) file [/var/lib/motioneye/Camera2/2017-03-12/13-05-54]: Operation now in progress

```

entmike, 2017-03-12:
I resolved this by cloning the Motion source and compiling motion 4.0.  The package that Ubuntu wanted to install was 3.2.  Hopefully this helps others who might run into it.

ccrisan, 2017-03-12:
Yeap, you need motion 4.0 for those things to work.

closed 2017-03-12 by ccrisan