motioneye-project/motioneye #491: Streaming a second camera kills all feeds

entmike, 2017-03-12:
I am running motioneye on Ubuntu.

First camera is a Network Camera pointing to a Rpi3 running MotionEyeOS that is running Fast Network Camera on default port 8081.

Ubuntu then is doing motion detection and further streaming out to port 9081 for another application (Home Assistant)

This works fine.

When I add a second camera on another server and set it up to stream out to 9082, i get this in error log and ALL feeds fritz out:

```
Mar 12 13:47:24 ubuntu meyectl[99062]:    ERROR: motion not running, starting it
Mar 12 13:47:27 ubuntu meyectl[99062]:    ERROR: connection problem detected for mjpg client for camera 2 on port 9081
Mar 12 13:47:27 ubuntu meyectl[99062]:    ERROR: connection problem detected for mjpg client for camera 1 on port 8082
Mar 12 13:47:27 ubuntu meyectl[99062]:  WARNING: Connect error on fd 15: ECONNREFUSED
```

After much troubleshooting, it seems that for reasons unknown, motioneye is confusing the hostname (or the port) of the streaming server and network camera...  Why else would it be trying to connect to cameras on 9081 and 9081, is why I think this.  Again, this only happens after adding a SECOND camera.

entmike, 2017-03-15:
FYI - Rebooting the server cleared this up.  I'm not sure what had its wires crossed.  Closing.

closed 2017-03-15 by entmike

fudgeboy2000, 2018-07-11:
Just wanted to comment on this one because I hit this exact issue last night, and it might help someone who lands here.  I've got 3 IP cameras with rtsp feeds, all of which work fine in VLC.  I could add any of them to motioneye on my Pi3 as the first camera and they would work, but as soon as I added any additional one, all the feeds would break.  A reboot didn't help me.

The problem came down to ports.  The motioneye logs gave this -  WARNING: mjpg client connection for camera 2 on port 8082 is closed

Each camera I added created a file like this -  /etc/motioneye/thread-1.conf.  The ports generated for each camera thread started at 8081 and go up, so thread-2 used 8082, thread-3 8083 and so on.  But 8082 was in use already on my Pi by node.  To fix it, I just had to manually edit the thread-* files to pick some free ports (use netstat -pln to see what's in use).  Changed the stream_port property and restarted motioneye with systemctl restart motioneye, and everything worked fine.