motioneye-project/motioneye #492: Stream starts and stops on Debian every view seconds

MrGlasspoole, 2017-03-13:
I can see the video for 2 seconds then i have the "no cam" picture for a view seconds.
This is constantly repeating.
```
Mar 12 19:52:57 motionEye meyectl[470]: WARNING: 403 GET /login/?_=1489344776452&_username=_&_signature=b9b208da54510d1b7d6690c149b4faead5a5258d (10.1.1.0) 1.28ms
Mar 12 19:53:02 motionEye meyectl[470]: ERROR: motion not running, starting it
Mar 12 19:53:02 motionEye motion: [0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Mar 12 19:53:02 motionEye motion: [0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
Mar 12 19:53:02 motionEye motion: [0] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started
Mar 12 19:53:02 motionEye motion: [0] [NTC] [ALL] motion_startup: Logging to syslog
Mar 12 19:53:02 motionEye motion: [0] [NTC] [ALL] motion_startup: Using default log type (ALL)
Mar 12 19:53:02 motionEye motion: [0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Mar 12 19:53:02 motionEye motion: [1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (http://10.1.0.11:8081/)
Mar 12 19:53:03 motionEye motion: [1] [ALR] [NET] netcam_handler_loop: Camera handler thread [2] started
Mar 12 19:53:03 motionEye motion: [1] [WRN] [NET] netcam_next: called with no data in buffer
Mar 12 19:53:03 motionEye motion: [1] [WRN] [NET] netcam_next: called with no data in buffer
Mar 12 19:58:14 motionEye motion: [1] [ALR] [NET] netcam_handler_loop: netcam camera handler: finish set, exiting
Mar 12 19:58:16 motionEye motion: [0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Mar 12 19:58:16 motionEye motion: [0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
Mar 12 19:58:16 motionEye motion: [0] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started
Mar 12 19:58:16 motionEye motion: [0] [NTC] [ALL] motion_startup: Logging to syslog
Mar 12 19:58:16 motionEye motion: [0] [NTC] [ALL] motion_startup: Using default log type (ALL)
Mar 12 19:58:16 motionEye motion: [0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Mar 12 19:58:16 motionEye motion: [1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (http://10.1.0.11:8081/)
Mar 12 19:58:16 motionEye motion: [1] [ALR] [NET] netcam_handler_loop: Camera handler thread [2] started
Mar 12 19:58:21 motionEye kernel: [ 6080.575975] motion[1144]: segfault at 7f82777ff150 ip 00007f82874c0e5f sp 00007f827d38e3f8 error 4 in libjpeg.so.62.1.0[7f8287487000+45000]
Mar 12 19:58:22 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:24 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:25 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:26 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:27 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:28 motionEye meyectl[470]: ERROR: motion not running, starting it
Mar 12 19:58:28 motionEye motion: [0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Mar 12 19:58:28 motionEye motion: [0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
Mar 12 19:58:28 motionEye motion: [0] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started
Mar 12 19:58:28 motionEye motion: [0] [NTC] [ALL] motion_startup: Logging to syslog
Mar 12 19:58:28 motionEye motion: [0] [NTC] [ALL] motion_startup: Using default log type (ALL)
Mar 12 19:58:28 motionEye motion: [0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Mar 12 19:58:28 motionEye motion: [1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (http://10.1.0.11:8081/)
Mar 12 19:58:28 motionEye motion: [1] [ALR] [NET] netcam_handler_loop: Camera handler thread [2] started
Mar 12 19:58:31 motionEye kernel: [ 6090.057404] motion[1148]: segfault at 7f530260b150 ip 00007f530e149e5f sp 00007f53040173f8 error 4 in libjpeg.so.62.1.0[7f530e110000+45000]
Mar 12 19:58:31 motionEye meyectl[470]: ERROR: connection problem detected for mjpg client for camera 1 on port 8081
Mar 12 19:58:31 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:32 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:34 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:35 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:36 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:37 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:38 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:40 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:40 motionEye meyectl[470]: ERROR: motion not running, starting it
Mar 12 19:58:40 motionEye motion: [0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Mar 12 19:58:40 motionEye motion: [0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
Mar 12 19:58:40 motionEye motion: [0] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started
Mar 12 19:58:40 motionEye motion: [0] [NTC] [ALL] motion_startup: Logging to syslog
Mar 12 19:58:40 motionEye motion: [0] [NTC] [ALL] motion_startup: Using default log type (ALL)
Mar 12 19:58:40 motionEye motion: [0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Mar 12 19:58:40 motionEye motion: [1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (http://10.1.0.11:8081/)
Mar 12 19:58:40 motionEye motion: [1] [ALR] [NET] netcam_handler_loop: Camera handler thread [2] started
Mar 12 19:58:40 motionEye motion: [1] [WRN] [NET] netcam_next: called with no data in buffer
Mar 12 19:58:40 motionEye motion: [1] [WRN] [NET] netcam_next: called with no data in buffer
Mar 12 19:58:42 motionEye kernel: [ 6101.897419] motion[1152]: segfault at 7f5a98afe150 ip 00007f5aa463ce5f sp 00007f5a9a50a3f8 error 4 in libjpeg.so.62.1.0[7f5aa4603000+45000]
Mar 12 19:58:43 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:44 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:45 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:46 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:47 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:48 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:49 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:50 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED
Mar 12 19:58:51 motionEye meyectl[470]: WARNING: Connect error on fd 15: ECONNREFUSED

```

ccrisan, 2017-03-13:
I guess at the time of writing ffmpeg hadn't made it to the backports. I'll update the docs. Can we close the issue now?

MrGlasspoole, 2017-03-13:
One thing:
I did install ffmpeg on another Debian from backports:

`~# echo "deb http://http.debian.net/debian jessie-backports main" | tee /etc/apt/sources.list.d/backports.list`

and there i had to install the codecs i needed (like libx264-dev and libopenjpeg-dev) separately.
So i guess if motionEye needs them you should take this into account.

Other then that it seems to run stable now so you can close it.