motioneye-project/motioneye #493: artifacts

snowboardrulez, 2017-03-16:
Hello, 

I'am quite new with motioneye but is running now for 1,5 months now on an RPI3 with 1 WansCam IP camera. I am very excited about the features of MotionEye and everything is running fine but sometimes i get some artifacts (I dont exacly how to explain) in the cameraview and pictures which are sent by mail.

MotionEye is running together with Domoticz & homebrigde on the same RPI3.
I already saw the problem when I only had MotionEye run on RPI3.

I already tried to lower the framerate and te resolution in Motioneye
The temperature is sometimes at 63 degrees

I attached a screenshot to explain the problem 
[image]

ccrisan, 2017-03-16:
This has nothing to do with motionEye but with the camera that you use or the way the motion daemon captures the stream from it. You should open an issue on the motion issue tracker. [this](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue) will help you report the issue.

closed 2017-03-16 by ccrisan

snowboardrulez, 2017-04-01:
So if i understand it right this problem is caused by the motion daemon service? Also when i use MotionEyeOS?

ccrisan, 2017-04-01:
Yes, it could be something that can be fixed by motion, or it could be a bug in your camera's firmware. Either way, it not something motionEye can fix. And the same applies to motionEyeOS (why would it be different?)