motioneye-project/motioneye #494: cross domain not allow

myB0B, 2017-03-17:
hi,
i receive a "cross-domain-not-allow" error when i'm trying to connect to stream from a web page hosted on another computer.
motioneye problem or motion problem ?

Mirio, 2017-07-12:
You can use the embed version to do that. You can generate it on "Video Streaming"->"Embed URL" and insert it  on the code below:
`<iframe height="360px" width="500px" src="<URL GENERATED BY MOTIONEYE>"></iframe>`

For example:
`<iframe height="360px" width="500px" src="https://127.0.0.1:6000/picture/60000/frame/"></iframe>`

elisman, 2018-01-23:
@myB0B this is not a motioneye problem and its not a motion problem. It is about pure web technologies. See [https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS](Cross-Origin Resource Sharing) for more details about your problem. And please close this issue as it not a problem of motioneye.

myB0B, 2018-01-25:
Sure, not a motioneye problem.
close

closed 2018-01-25 by myB0B

Bluscream, 2023-06-25:
omg of course it is a motioneye problem because motioneye sets the headers for its webserver ...

myB0B, 2023-06-26:
haha, a little too late, but yes this is a motioneye problem, python webserver problem if i remember, i did it differently at the time....

michaelachrisco, 2024-04-05:
I came across this issue as well. I have my domain setup correctly, everything else came through. Im still getting cors errors specifically on the cameras themselves.

zagrim, 2024-04-05:
If (and that is a rather big 'if' unfortunately) I'm understanding this correctly, we'd need to have ME add `Access-Control-Allow-Origin: *` to the responses in order for the embedding of the UI to be possible. Am I correct?

I do fail why someone would want to embed whole ME in another site, and I assume only a few users would have such a requirement. I seems there might be some security implications allowing that, but given https://security.stackexchange.com/questions/43639/why-is-the-access-control-allow-origin-header-necessary they might be rather limited. But AFAIK ME is also used in some companies, where the threat landscape might differ from the home users.

Given that camera stream works currently by continuously sending requests for current frame _with credentials in query parameters_ that at least would need some more inspection regarding whether that would open any vulnerabilities if ACAO was set to allow all origins (cookie-based authentication should be ok I guess).

Anyway, if this was to be made possible, it should definitely be disabled by default. If someone can provide a plausible use-case for this maybe we should re-open this issue.

Bluscream, 2024-04-06:
> If (and that is a rather big 'if' unfortunately) I'm understanding this correctly, we'd need to have ME add `Access-Control-Allow-Origin: *` to the responses in order for the embedding of the UI to [...]

HomeAssistant is my usecase. It allows embedding the ME page or single streams into its dashboard to view and navigate to and from it with touchcontrols wwhile retaining its sidebar and not needing embedded browser controls