motioneye-project/motioneye #495: Problem DIGOO rtsp Camera

aspirtakis, 2017-03-21:
Dear Motion ! i have installed motionEy OS and is working great with some usb cameras on my raspberry pi 2 
On my Try to add RTSP link that works fine on VLC i get HEader Error 
the link  i have on vlc is 
rtsp://admin:224935@192.168.2.104/onvif1 
this works on VLC 
I tried added on motion with no luck and is the 2nd RTSP camera that i cannot ADD 

the log show this 
Traceback (most recent call last):
  File "home/ccrisan/motionpie/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/curl_httpclient.py", line 446, in _curl_header_callback
  File "home/ccrisan/motionpie/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/httputil.py", line 189, in parse_line
ValueError: need more than 1 value to unpack

as i read Mr DAve is included on the OS so 
is there anything i do wrong ?
 i can provide remote access to PI if this Helps

carryonrewardless, 2017-04-15:
I have my Digoo M1Q setup on MotionEye like rtsp://192.168.2.104/onvif2 (note not onvif1) using your link. From memory the username and pwd were just entered in the fields when setting up. Check your password. Unless you have changed it the default admin pwd is 20160404.
Hope that helps.

wouterkabouter77, 2017-04-26:
When you add the camera you must choose the option rtsp/udp stream in the checkbox camera. See the screenshot below. 

[image]

aspirtakis, 2017-04-26:
Its working with the above Settings /onvif2 
it works also with onvif1 but the PI crash imidatelly 

thanks ALOT for the help

jjpeet, 2019-01-01:
Hi @ccrisan I also have the same problem with these cameras.
I have been running them directly into Home Assiatnt for a number of years without a problem; using the FFMPEG component. Also, I’m able to open the cameras using VLC.

If I use the approach mentioned above with MotionEye, sometimes a camera image will appear after about 15-30min and then won’t appear after a restart for the same amount of time or worse.

This is patchy, and works with only some of my Digoo M1Q cameras, and others of the same model/firmware not at all. I have other cameras which come into MotionEye without a problem.

I am running MotionEyeOS on a Pi3 (and also as an add-on within HASSIO on a seperate Pi3).

Is there any way I can get these Digoo cams to run reliably in MotionEye?

Thanks for your work on this project its very impressive...

Thx
James

JamBalaya56562, 2026-11-07:
Closing as solved, as you confirmed that the camera works with the settings above: the `/onvif2` URL and the RTSP/UDP option. If a Digoo camera only shows a picture now and then, please open a new issue with the motionEye log.

closed 2026-11-07 by JamBalaya56562