motioneye-project/motioneye #497: "no camera" found RPi Zero W

cephasiii, 2017-03-23:
Greetings,
     I am having an issue getting motioneye to recoqnize the camera. I have setup two other RPi's without issue. I ran the "raspistill -o cam.jpg" with success. Also, "bcm2835-v4l2" is added to "/etc/modules". 

     Any suggestions?
[image]

wagsforever, 2017-03-23:
I have the same issue. On the motioneyeos page the developer said the Zero W is not supported yet so I assume it's the same issue for the motioneye

cephasiii, 2017-03-23:
Thanks. I just installed motioneye/os on another Zero W without issue. I tried the camera on a Zero and still can't find the camera. I think I narrowed it down to a software issue, but its weird because, like I said, I installed everything on another Zero W without any issues.

wagsforever, 2017-03-23:
Let me know if you work it out because, as I say, I have the same issue.

cephasiii, 2017-03-25:
Bump... Anyone have a solution for this issue?

ccrisan, 2017-03-26:
Have you checked this troubleshooting article? https://github.com/ccrisan/motioneye/wiki/Troubleshooting#raspberry-pi-camera-module-not-detected

Jay-Arr, 2017-03-27:
One point of contention: if you're using a Raspberry Pi Zero W with a camera module, and the new official Pi Foundation case, check to make sure the connector for the camera sensor is attached on the board correctly.  If you're not careful, when you put the camera module into the lid of the case, the connector on the front of the module's board that comes from the camera sensor and lens can get tilted and disconnect, and so the software will not detect the module at all, even if the ribbon connector to the Pi is connected correctly.  That's what happened to me.  Take the module out, and check to make sure the connector on the front is straight and level.  Even disconnect it and reconnect it to make sure it's on there properly.  I went back and forth between the motioneye on Raspbian, and motioneyeos, trying to figure out why camera wasn't working.  Finally settled on the os version once I figured out it was this odd hardware issue.  Works great now!

Jay-Arr, 2017-03-27:
Here's a pic of what I'm talking about:
[image]

aptonline, 2017-03-27:
Refered to as the Sunny connector I believe.

Jay-Arr, 2017-03-27:
The original connector was yellow.

cephasiii, 2017-03-28:
I will check the connector, thanks.

wagsforever, 2017-03-28:
Hi Jay-Arr. I'm still having no camera showing up even after updating motioneye. Did your camera work with raspistill before you fixed the issue with the connector?

Jay-Arr, 2017-03-28:
No.  But once I fixed the connector it did.  The circuitry for the camera is right in the lens module on the camera board.  If the above circled connector isn't flush and level, the computer won't see the camera module at all, regardless of the state of the external ribbon cable that goes to the Pi.  The board just has a few resistors for power and such.

wagsforever, 2017-03-28:
See my camera does work with raspistill so this can't be the issue. Annoying. Thanks anyway

Jay-Arr, 2017-03-28:
Have you tried the camera with the new March build of motioneyeos?  See if that works.  That's what I'm using now.

wagsforever, 2017-03-28:
I'll give it a go. Cheers

Jay-Arr, 2017-03-28:
You won't need to change the /etc/modules like you do on Raspbian.  You WILL need to use a USB Ethernet port to get network access from the start (the system reboots continually if it doesn't find a network connection on boot), but that's only temporary.  Wireless settings can be configured remotely through the web interface, and then you're good, but you can at least test the camera through wired Ethernet first, since that's your problem.  You can alternatively create a wpa_supplicant.conf file with the wireless connection settings.  

It is an embedded-style Linux build though, so you'll have limited command-line utilities if you do try to use a local console.  If you're just looking to use the included features, it's probably better to use the OS version for simplicity's sake, but YMMV.

soydepr, 2017-04-02:
i have same issue on pi 3 i can see camera with raspistill but motion eye does not see it

agilob, 2017-05-02:
No longer an issue on raspi 0 W, works out of box.

closed 2019-06-21 by ccrisan