motioneye-project/motioneye #500: Record Audio along with Video

Jo-Pitts, 2017-03-27:
Hi there.  I've installed motioneye on my Pi 3.  It works like a dream, but I have one requirement that I've not yet met.

I need to record audio whenever I am recording a motion capture video.  I don't need to stream this to the browser though (nice if I can, but not needed).

Any thoughts on how to do this?  I believe ffmpeg can do this, so maybe a config option to 
a. turn on/off audio recording
b. select the audio device to match to a camera
c. audio quality

that is probably it?

please help?  I'll even pay a small bounty if needed

ccrisan, 2017-03-28:
Audio recording should normally be implemented in motion (which is responsible for the video recording). As far as I know, they don't plan to add support for audio, so the answer is no, for now.

closed 2017-03-28 by ccrisan

Jo-Pitts, 2017-03-29:
Ahhh... I had assumed you were just using Motion to detect the motion, and
then using the motion on_event_start to hand off to ffmpeg for recording
and then on_event_end to tell ffmpeg to stop.

I'll try to home brew something using that approach :)

Cheers
Jo