motioneye-project/motioneye #501: Monitoring Commands from GPIO pins

lebo202002, 2017-03-28:
I can't figure out how to write a bash script to enable an overlay button/icon (on the camera view) to change colors when one of the GPIO pins (setup as an input pin) changes states (either from high to low or low to high). 

I only found one example script here https://github.com/ccrisan/motioneye/wiki/Monitoring-Commands
but wasn't able to backwards engineer it for my intended purposes. 

I'm hoping to use this to tell when a level switch goes hi/low or when a CO detectors goes off or when a door is opened. 

Any help is much appreciated.

causalityzach, 2017-08-06:
I'm not sure if you have figured it out, but hopefully this will at least help someone in the future. https://bitbucket.org/MattHawkinsUK/rpispy-misc/raw/master/motionEyeOS/monitor_1

lebo202002, 2017-08-06:
> MattHawkinsUK/rpispy-misc/raw/master/motionEyeOS/monitor_1

Cz

no. I abandoned the idea as I'm WAY more of a hardware guy than a software
guy. But I believe I can use this to pick up where I left off.

Thanks again for responding to my request!

JamBalaya56562, 2026-11-07:
Closing as answered, as the script linked above shows how to do this and you said you can pick up from there. Thanks for following up.

closed 2026-11-07 by JamBalaya56562