motioneye-project/motioneye #502: Motioneye deleted GBs of media files from my NAS!

Muzurio, 2017-03-28:
I just found out that last week Motioneye has deleted almost 350 GB of movies and mp3 from my NAS (synology)!.

How could this have happened? . I had Motioneye installed to write to a certain share (/Media/Cameras) and later changed this to another share. Looking at the logs of synology I now see that Motioneye started deleteing all the files in /Media . I have lost years of movies now .......

ccrisan, 2017-03-28:
There's a big warning when you enable limited preservation of media files. It says that motionEye will delete all media files it finds in the root of the camera folder. You should never set the camera root folder to the same folder you keep your personal movie files. Instead, always use something specific to the camera.

closed 2017-03-28 by ccrisan

Muzurio, 2017-03-28:
That is not a sound design. Motioneye knows which folders it made, why would it remove files in other folders. I just found out it was even worst. Motioneye deleted 1,5 TB :(((((((((

ccrisan, 2017-03-28:
motionEye does not know which folders it created. It only knows which is the current root folder of a camera. If you change the folder (like you mentioned you did), it has no way to know you pointed it to a folder where you store your other files.