motioneye-project/motioneye #504: motioneye on Raspian-Jesse

pgas37, 2017-04-02:
according instructions:
http://www.howtoembed.com/projects/raspberry-pi/95-motioneye-with-raspberry-pi
i installed the "motioneyos-20170212.tar.gz"" to / home/pi/motioneye.

It should be possible to start motioneye with the command "./motioneye.py", but such a file is not available.

What is the command to start motioneye manually?

thanks for the help

ccrisan, 2017-04-02:
That is an old article on motionEye. You have downloaded [motionEyeOS](https://github.com/ccrisan/motioneyeos). Follow [these](https://github.com/ccrisan/motioneyeos/wiki/Installation) instructions.

ccrisan, 2017-04-02:
Also if you really want plain motionEye on Raspbian, see [this](https://github.com/ccrisan/motioneye/wiki/Installation).

closed 2017-04-02 by ccrisan