motioneye-project/motioneye #506: AUDIO+VIDEO save in motioneye

pallaviawate, 2017-04-05:
Hello,
 great tutorial,it works fine to me.actully using this i can save only motion,but i want to save audio+video using motioneye. what can i do for this to save audio+video using motioneye??

MrDornback, 2018-11-28:
I also am interested in the ability to connect a simple audio sensor (from a Pi pack for example) and record sound an video.

ilovelinux, 2019-06-20:
See https://github.com/ccrisan/motioneyeos/issues/1170 https://github.com/ccrisan/motioneyeos/issues/1322 https://github.com/ccrisan/motioneye/issues/1112 https://github.com/ccrisan/motioneyeos/issues/105 https://github.com/ccrisan/motioneyeos/issues/1512
Workaround, since Motion isn't supposed to record audio: https://www.lavrsen.dk/foswiki/bin/view/Motion/SoundAudioRecording

ccrisan, 2019-06-21:
We don't have any plans to add audio recording support to motionEye, especially given that Motion doesn't support it either.

closed 2019-06-21 by ccrisan

CLUB777ONE, 2019-07-11:
What's wrong with you? For years, I always read the same answer. - no plans, no, no, no, - As if audio recording was a sin. ... If many users have the same wish, then this should be implemented. Otherwise someday somebody will fulfill the wishes of your fans and motionOS will fall into oblivion.

jasaw, 2019-07-12:
@CLUB777ONE MotionEye(OS) relies on [motion](https://github.com/Motion-Project/motion) to record video. If we want audio recorded in the movie, motion software needs to support that first. Can you please convince motion developers to support audio recording?

ilovelinux, 2019-07-12:
It can be done in post as it's written in the link I sent. Motion is not going to implement audio because no secutiry camera has audio integrated and there are working workaround to add audio to recording. I think that you can easly use extpipe too.

Maybe MotionEye could show a checkbox to enable audio recording and automatically configure motion to do so

CRImier, 2019-07-12:
@CLUB777ONE As most cameras don't even have audio, motioneye is not going to lose any significant portion of its userbase even after such a fork appears.

@ACEDornback that won't work, you do need a sound card with an input to record audio, a "microphone" sensor" from a sensor pack won't work (esp. given the majority of these are not amplifiers and are merely audio level triggers).

I think that some external software could be more-or-less easily written - one that receives motion start/stop events from motioneyeos (these are already available) and records audio from an audio source, then joins the video recording file together with the audio. I imagine that timing mismatches between audio and video are inevitable, though.

DeadEnded, 2020-05-30:
Not to Necro, but I am using a script to hack the audio into the video recordings.
It's probably not the cleanest, but it works.

wokawoka, 2021-10-31:
@DeadEnded I'm necroing your necro... I'm very interested in the script you are using. 
Would you mind sharing it? I'm sure many other people, like myself, would find it extremely useful. Thank you!

DeadEnded, 2021-11-02:
@wokawoka the script is in a repo on my profile... but I haven't used MotionEye in over a year now...
I've moved to a program called Frigate... and it's just awesome.
It's built for object detection - not just motion - and it works really well.
But I don't want to blab about it on here - if you want to look at the script check out the repo.
I'll try to keep it maintained, but I don't use it - so might be best to fork it and customize as needed.

Cheers!
DeadEnd

wokawoka, 2021-11-02:
@DeadEnded 
great, I've just found your repo and I will check the script as soon as I'll have some time. 
Thank you very much!