motioneye-project/motioneye #510: Unable to find lastsnap.jpg

cbarratt, 2017-04-10:
My setup is a rPi 3 with rPi Camera v2 and running DietPi v148.

Motioneye was installed via the DietPi software installer GUI.

I'm currently struggling to see how or where to configure the last snap shot image. afaik it should be already enabled?

```
motionEye Version	0.35.2
Motion Version 4.0.1
```

Still images are enabled on `motion detection` with the preserve setting to `forever` 

Images do get outputted and everything seems ok. 

An example of the folder structure:

```
root@DietPi:/mnt/dietpi_userdata/motioneye# ls -l
total 4
drwxr-xr-x 3 root root 4096 Apr 10 12:28 Camera1
root@DietPi:/mnt/dietpi_userdata/motioneye# cd Camera1/
root@DietPi:/mnt/dietpi_userdata/motioneye/Camera1# ls -la
total 12
drwxr-xr-x 3 root root 4096 Apr 10 12:28 .
drwxrwxr-x 3 root root 4096 Apr  9 22:40 ..
drwxr-xr-x 2 root root 4096 Apr 10 12:28 2017-04-10
-rw-r--r-- 1 root root    0 Apr 10 10:17 .keep
root@DietPi:/mnt/dietpi_userdata/motioneye/Camera1# cd 2017-04-10/
root@DietPi:/mnt/dietpi_userdata/motioneye/Camera1/2017-04-10# ls -la
total 4276
drwxr-xr-x 2 root root   4096 Apr 10 12:28 .
drwxr-xr-x 3 root root   4096 Apr 10 12:28 ..
-rw-r--r-- 1 root root 395312 Apr 10 12:28 12-28-11.jpg
-rw-r--r-- 1 root root 396095 Apr 10 12:28 12-28-12.jpg
-rw-r--r-- 1 root root 396482 Apr 10 12:28 12-28-13.jpg
-rw-r--r-- 1 root root 396049 Apr 10 12:28 12-28-14.jpg
-rw-r--r-- 1 root root 395871 Apr 10 12:28 12-28-15.jpg
-rw-r--r-- 1 root root 396648 Apr 10 12:28 12-28-16.jpg
-rw-r--r-- 1 root root 396383 Apr 10 12:28 12-28-17.jpg
-rw-r--r-- 1 root root 396284 Apr 10 12:28 12-28-18.jpg
-rw-r--r-- 1 root root 396285 Apr 10 12:28 12-28-19.jpg
-rw-r--r-- 1 root root 396040 Apr 10 12:28 12-28-20.jpg
-rw-r--r-- 1 root root 396182 Apr 10 12:28 12-28-21.jpg
root@DietPi:/mnt/dietpi_userdata/motioneye/Camera1/2017-04-10#`
```

imperialjoy, 2017-04-23:
I have the same issue, I need the lastsnap.jpg so it can integrate with homebridge and send the image to my devices.

DaniW42, 2020-07-26:
I can confirm that lastsnap.jpg is not created on my Raspberry Pi

**motionEye Version** 0.42.1
**Motion Version** 4.2.2
**OS Version** Raspbian 10

Is there any way to fix this? I had another motioneye setup up and running on a vm and it created a symlink to the latest file called lastsnap.jpg (as intended).

[image]