motioneye-project/motioneye #511: Button script

Ramon-87, 2017-04-12:
Hello,

I'm trying to program my buttons. But I can't figure out how it works.

The link for PTZ turning left is .../cgi/ptdc.cgi?command=set_relative_pos&posX=-10&posY=0

I was thinking that the curl script should be:

```
#!/bin/bash

curl "http://.../cgi/ptdc.cgi?command=set_relative_pos&posX=-10&posY=0
```
or like the Wiki says:

```
#!/bin/bash

URL="http://.../cgi/ptdc.cgi?command=set_relative_pos&posX=-10&posY=0"
METHOD="POST"
TIMEOUT="5"
curl -X $METHOD --connect-timeout $TIMEOUT "$URL" > /dev/null
```

But there is no respons in MotionEyeOS.

In advance: Thanks for helping!

ccrisan, 2017-04-13:
Try executing the script manually and make sure it works. From what I can see, there are no closing quotes `"`. Also make sure it's executable (use `chmod +x` for that).

Ramon-87, 2017-04-18:
Thank you for you're reply.

Indeed in this example I forgot the closing qoutes. I used them in my script. I've used `chmod +x`. 
Even when I execute it manual the camera doesn't move. But it gave me an permission error, which I fixed with: `User:Password@....`.

Still nothing and I can't figure out what I'm doing wrong.